particle-iot/device-os #489: A newer version of gcc-arm-none-eabi-49 is required that the one that brew downloads

osmithy, 2015-07-15:
I followed all the steps [here][https://github.com/spark/firmware/blob/latest/docs/dependencies.md#1-download-and-install-dependencies] to get the dependencies downloaded, but I'm getting this error:

/firmware/build/arm-tools.mk:42: *** "ARM gcc version 4.9.3 20150528 or later required, but found 4.9.3 20150303 (release) [ARM/embedded-4_9-branch revision 221220]".  Stop.

I'm on OSX. Anyone know what might be going wrong?

dmiddlecamp, 2015-07-15:
I think it looks like you have "20150303", but it needs "20150528", can you double-check to make sure you grabbed the latest one from the gcc arm embedded site?  sorry!

joegoggins, 2015-07-15:
I just tried the homebrew instructions for OS X and got the same thing. Homebrew installs an older package. I issued a patch + pull request to homebrew to fix this https://github.com/PX4/homebrew-px4/pull/7.

In the meantime, before that gets merged. You can `brew edit gcc-arm-none-eabi-49`, and apply the changes i made in that PR, brew uninstall, and brew reinstall.

osmithy, 2015-07-15:
Thanks man. I didn't realize you could edit formulas, and I'm pretty new to github, so I fixed it by copying the repository to my own github account and making the changes there, probably right as you were making them yourself. So it works! But now I'm getting errors trying to compile/install the latest "develop" branch of the firmware because a fix just got made for this [I2C bug](https://community.particle.io/t/issue-with-i2c-on-photon/12161/25) that I'm running into. I'm going to ask on that thread just to keep things organized - feel free to answer there if you have any idea what's going on!

joegoggins, 2015-07-15:
For others stumbling on this thread, something IMPORTANT + nuanced to watch out for: If you ran `make` before on an old compiler, upgraded, then ran `make` again. Things will probably blow up in weird ways. 

To fix, either run `make clean` or nuke the whole repo and clone from scratch to make that weirdness go away.

joegoggins, 2015-07-15:
Dang. Open source moves fast. That PR  already got merged above. Closing this issue.

closed 2015-07-15 by joegoggins

towynlin, 2015-07-15:
:+1: :100: :smiley_cat: :trophy: