particle-iot/device-os #490: client.connect(hostname, port) causes SOS if DNS not resolved

pkourany, 2015-07-16:
On latest develop and 0.4.3 branches, running client.connect(hostname,port) where the DNS is not resolving (in my case perhaps because of a VPN), the function will cause an SOS and the Photon resets.

HardWater, 2015-07-17:
comments copied from the community site
First Posting 
-----------------------------------------------------------------------------------------------------------------
@peekay123 I am continuing to drill down to find out where this problem is. What I have found out so far is the problem occurs at this call in spark_wiring_tcpclient.cpp

       uint32_t ot = HAL_WLAN_SetNetWatchDog(S2M(MAX_SEC_WAIT_CONNECT));
       DEBUG("_sock %d connect",_sock);
       //gets here first and second request
       connected = (socket_connect(_sock, &tSocketAddr, sizeof(tSocketAddr)) == 0 ? 1 : 0);
       DEBUG("_sock %d connected=%d",_sock, connected);
       //gets here only on first request on the second request WDT seems to trip before here

@mdma is this something that you would be looking into or is there someone else that should be pinged?
-----------------------------------------------------------------------------------------------------------------
Second Posting

@mdma Making further progress on finding where the problem happens here is part of socket_hal.cpp

sock_result_t socket_connect(sock_handle_t sd, const sockaddr_t *addr, long addrlen)
{
    sock_result_t result = SOCKET_INVALID;
    socket_t* socket = from_handle(sd);
    if (is_tcp(socket)) {
        wiced_result_t wiced_result = wiced_tcp_bind(tcp(socket), WICED_ANY_PORT);
        if (wiced_result==WICED_SUCCESS) {
            wiced_tcp_register_callbacks(tcp(socket), socket_t::notify_connected, socket_t::notify_received, socket_t::notify_disconnected, (void*)socket);
            SOCKADDR_TO_PORT_AND_IPADDR(addr, addr_data, port, ip_addr);
            unsigned timeout = 5*1000;
           //-----------> gets here on first and second request
            result = wiced_tcp_connect(tcp(socket), &ip_addr, port, timeout);
           //-----------> gets here only on the first request WDT reset before here on the second
        }

        if (result!=WICED_SUCCESS)
            {
            result = as_sock_result(wiced_result);
            }
    }
    return result;
}

I do not seem to be able to get closer to the problem as I cannot find the source for wiced_tcp_connect() I am assuming that is part of the WICED binary image. Please Help?

Third posting
-----------------------------------------------------------------------------------------------------------------
I believe the problem lies at the lowest levels of setting and closing tcp connections. The other idea of what it could be is the the WDT is not set correctly second time around.

HardWater, 2015-07-17:
Just confirmed this problem **does not** exist with the photon_042 branch. Which is good news that it has not always existed but bad I need other features that are part of photon_043.  I think @peekay123 alludes to this fact that this was not a problem before. Just thought I would check for sure.

mtnbrit, 2015-07-20:
ill add a +1 to this, in my case I have to POST to several hosts in a list, my loop gets past the first 2 but does this WDT reboot on the 3rd connect. Also I'm not doing DNS, just a straight connect to the raw IP and port. So perhaps we can eliminate DNS as being related.

m-mcgowan, 2015-07-20:
This is bizzare - the watchdog is supposed to be disabled. @mtnbrit - do you have some code to share that I can try out to reproduce the problem, ideally a minimal example of the problem would b great to have! Thanks.

mtnbrit, 2015-07-20:
my code and environment is very specific and you would have trouble replicating it, i.e. the devices the photon needs to post to are on my local lan only.

this is code that worked fine on the Core by the way.

is the sos one flash necessarily a WDT issue?

m-mcgowan, 2015-07-20:
SOS 1 is a hard fault. I don't know if it's actually a watchdog issue - just quoting your previous post (and @HardWater's comment above, https://github.com/spark/firmware/issues/490#issuecomment-122333498)

I will try some various TCP test apps to try to replicate the failure.

kneuron, 2015-08-06:
we seem to be having similar issues. often SOS on client.connect() Photon, 0.4.3

HardWater, 2015-08-06:
Kneuron give the develop branch a try. Matt made a change a couple weeks back that fixed the problem for me.

monkbroc, 2015-08-07:
Here are steps to reproduce the problem:

* Flash latest develop firmware (any user application will do)
* Connect the Photon to the internet
* Modify router settings to block all ports for the Photon MAC address
* Reset Photon
* SOS after 5 seconds of blinking cyan (SOS in `wiced_tcp_delete_socket` or `wiced_tcp_disconnect`)

Here are the settings in my router:
[image]

pkourany, 2015-08-07:
@AndyW did some sleuthing on this here:

https://community.particle.io/t/long-term-photon-connection-stability/13925/36

javier-rodas, 2015-08-07:
Hello everybody @pkourany @m-mcgowan 

The issue also happens when connecting to an IP address (without using DNS), as it was already reported in #516. I have tested it with the Photon develop firmware versions **0.4.3** and **0.4.4-rc2**.

I have discovered that the ```client.connect(server, port))``` call **ALWAYS returns 1**, even when the **port is closed** at the server side.

Let's see some testing results.

With the following first two pieces of code the Photon **DOES NOT PANIC** and the **network call doesn't fail** in the case the remote server port is closed:

**CODE1 (AUTOMATIC):**

```
#include "application.h"

TCPClient client;

IPAddress server(192,168,0,101);
int port = 8888;

SYSTEM_MODE(AUTOMATIC);

void setup() {
	Serial.begin(9600);

	delay(1000);

	Serial.println("connecting...");
	if (client.connect(server, port)) {
    	Serial.println("connected");
		client.println("HELLO WORLD!");
	} else {
		Serial.println("connection failed");
	}
}

void loop() {}
```

**CODE2 (SEMI_AUTOMATIC)**

```
#include "application.h"

TCPClient client;

IPAddress server(192,168,0,101);
int port = 8888;

SYSTEM_MODE(SEMI_AUTOMATIC);

void setup() {
	Serial.begin(9600);

	WiFi.connect();
	delay(1000);

	Serial.println("connecting...");
	if (client.connect(server, port)) {
    	Serial.println("connected");
		client.println("HELLO WORLD!");
	} else {
		Serial.println("connection failed");
	}
}

void loop() {}
```

In both cases the result is the same, ```client.connect(server, port)``` always return 1, enters in the "connected" if branch, and the ```client.println("HELLO WORLD!");``` call NEVER fails even with the remote server port closed. **BUT, when the remote server port is open, the if branch executed is the same and the TCP server receives the "HELLO WORLD!" successfully!!.**

[image]
**Figure 1. CoolTerm screen capture with the remote server port closed**

---

**CASE 3:** Moving the client.connect to the loop() code and produce some kind of network "overflow" but NEVER PANICS:

```
#include "application.h"

TCPClient client;

IPAddress server(192,168,0,101);
int port = 8888;
int i=0;
//SYSTEM_MODE(SEMI_AUTOMATIC);
SYSTEM_MODE(AUTOMATIC);

void setup() {
	Serial.begin(9600);

//	WiFi.connect();
	delay(1000);
}

void loop() {
	Serial.println("connecting..." + String(i));
	if (client.connect(server, port)) {
    	Serial.println("connected");
		client.println("HELLO WORLD!");
	} else {
		Serial.println("connection failed");
	}
	i++;
}
```

Output with the server port CLOSED:
```
<---- The Photon connects to the Wifi (green) and correctly to the Cloud (breath cyan)
connecting...0       <---- Here it seems that it tries to connect, becase the terminal output stops for a short while, and then it enters in the "connected" if branch and then the Serial output is really fast
connected
connecting...1
connected
...
connecting...765
connected
connecting...766  <--- After 765 iterations the socket client connection starts to fail!!!, as it should be from the beginning as the server port was always closed.
connection failed
connecting...767
connection failed
connecting...768
connection failed
...
connecting...6160  <---- After about ~6150 iterations the Photon finally disconnects from the cloud, and starts blinking cyan
connection failed
connecting...6161
connection failed
connecting...6162
connection failed
```

---

**CASE 4:** We can reproduce a PANIC after calling ```client.connected()``` when this incorrect "connect" situation happened when the server port is closed => Uncomment the latest line to produce the PANIC.

```
#include "application.h"

TCPClient client;

IPAddress server(192,168,0,101);
int port = 8888;

//SYSTEM_MODE(SEMI_AUTOMATIC);
SYSTEM_MODE(AUTOMATIC);

void setup() {
	Serial.begin(9600);

//	WiFi.connect();
	delay(1000);

	Serial.println("connecting...");
	if (client.connect(server, port)) {
    	Serial.println("connected");
		client.println("HELLO WORLD!");
	} else {
		Serial.println("connection failed");
	}
}

void loop() {
    Serial.println("Whatever...");
//    Serial.println("connected: " + String(client.connected()));
}
```

[image]
**Figure 2. Output without uncommenting the latest sentence => DO NOT PANIC:**

[image]
**Figure 3. Output after uncommenting the latest sentence  => DO PANIC:****

NOTE: These examples were tested with the 0.4.4-rc2 firmware version.

@m-mcgowan  Hope this helps you to identify the issue in the latest Photon develop firmware versions 0.4.3+

andyw-lala, 2015-08-07:
Nice work triangulating on this.

monkbroc, 2015-08-07:
@javier-rodas 
> I have discovered that the client.connect(server, port)) call ALWAYS returns 1, even when the port is closed at the server side.

I made a pull request #519 to fix that bug, but the SOS still remains (instead of crashing on `wiced_tcp_disconnect` it crashes on `wiced_tcp_delete_socket`).

andyw-lala, 2015-08-08:
Please try the workaround I mention here:
https://community.particle.io/t/long-term-photon-connection-stability/13925/48

It boils down to this 3 line patch:

diff --git a/hal/src/photon/inet_hal.cpp b/hal/src/photon/inet_hal.cpp
index b42812b..418e3b0 100644
--- a/hal/src/photon/inet_hal.cpp
+++ b/hal/src/photon/inet_hal.cpp
@@ -31,7 +31,9 @@ int inet_gethostbyname(const char* hostname, uint16_t
hostnameLen, HAL_IPAddress
     wiced_ip_address_t address;
     address.version = WICED_IPV4;
     wiced_result_t result = wiced_hostname_lookup (hostname, &address,
5000);
-    out_ip_addr->ipv4 = GET_IPV4_ADDRESS(address);
+    if (result == WICED_SUCCESS) {
+        out_ip_addr->ipv4 = GET_IPV4_ADDRESS(address);
+    }
     return -result;
 }

[also attached, because gmail does awful things to patches]

This will only help is the client.connect() code is doing the same
brain-dead thing and ignoring function return values, and relying on
whether *ipaddr == 0.0.0.0 instead.

I say workaround, because I believe all callers of inet_gethostbyname()
should check the return value, not *ipaddr.

andyw-lala, 2015-08-08:
OK - I took a look at spark_wiring_tcpclient.cpp and it looks like the
return value of inet_gethostbyname() is not ignored, but it not passed back
to the caller either..

The attached patch should make things behave better.

Please test and report back. I can do a pull request if testing proves it
worthwhile.

andyw-lala, 2015-08-08:
You are going to tell me that github strips out attachments, aren't you...

Here's the client_connect patch:

diff --git a/wiring/src/spark_wiring_tcpclient.cpp
b/wiring/src/spark_wiring_tcpclient.cpp
index badfcf4..16adda0 100644
--- a/wiring/src/spark_wiring_tcpclient.cpp
+++ b/wiring/src/spark_wiring_tcpclient.cpp
@@ -58,7 +58,7 @@ int TCPClient::connect(const char* host, uint16_t port,
network_interface_t nif)
       {
         IPAddress ip_addr;

-        if(inet_gethostbyname(host, strlen(host), ip_addr, nif, NULL) == 0)
+        if((rv = inet_gethostbyname(host, strlen(host), ip_addr, nif,
NULL)) == 0)
         {
                 return connect(ip_addr, port, nif);
         }

javier-rodas, 2015-08-08:
More testing here. This time for testing the **TCPClient** ```connect()``` and ```connected()``` methods I used the the **REST Client Library** available at [spark-restclient][1], with the Photon (and the latest develop firmware versions), which uses the TCPClient library behind:

I set ```#define HTTP_DEBUG``` in the ```rest_client.cpp``` file for debugging purposes.

Testing ```application.cpp``` code:
```
#include "rest_client.h"

RestClient client = RestClient("192.168.0.101", 8080);

String response;
int statusCode;

void setup() {
  Serial.begin(9600);
  Serial.println("starting...");
  
  while(!Serial.available()) Spark.process();
}

void loop() {
    response = "";
    statusCode = client.post("/", "parameter1=123&parameter2=555", &response);
    Serial.println("Status code from server: " + String(statusCode));
    Serial.println("Response body from server: " + response);
    delay(1000);
}
```

Output:

```
starting...
HTTP: connect
HTTP: connected
REQUEST: 
POST / HTTP/1.1
Host: 192.168.0.101
Connection: close
Content-Length: 29
Content-Type: application/x-www-form-urlencoded

parameter1=123&parameter2=555

HTTP: call readResponse
HTTP: NON-NULL RESPONSE POINTER: 
HTTP: RESPONSE: 
.,H.,T.,T.,P.,/.,1.,..,1., .,2.,0.,0., .,O.,K.,
.,
.,X.,-.,P.,o.,w.,e.,r.,e.,d.,-.,B.,y.,:., .,E.,x.,p.,r.,e.,s.,s.,
.,
.,C.,o.,n.,t.,e.,n.,t.,-.,T.,y.,p.,e.,:., .,a.,p.,p.,l.,i.,c.,a.,t.,i.,o.,n.,/.,j.,s.,o.,n.,;., .,c.,h.,a.,r.,s.,e.,t.,=.,u.,t.,f.,-.,8.,
.,
.,C.,o.,n.,t.,e.,n.,t.,-.,L.,e.,n.,g.,t.,h.,:., .,1.,5.,
.,
.,D.,a.,t.,e.,:., .,S.,a.,t.,,., .,0.,8., .,A.,u.,g., .,2.,0.,1.,5., .,1.,7.,:.,4.,1.,:.,0.,7., .,G.,M.,T.,
.,
.,C.,o.,n.,n.,e.,c.,t.,i.,o.,n.,:., .,c.,l.,o.,s.,e.,
.,
.,
.,
.,{.,".,s.,t.,a.,t.,u.,s.,".,:.,".,o.,k.,".,}...................................................................................................................
................................................................................................................................................................
................................................................................................................................................................
................................................................................................................................................................
................................................................................................................................................................
................................................................................................................................................................
................................................................................................................................................................
................................................................................................................................................................
................................................................   <--- infinite output
```

The problem here is the **the TCPCliente connected() function doesn't detect when the socket is closed** (at the server side), and **it will always return 1**, so that the following ```while (client.connected())``` loop defined in ```rest_client.cpp``` never ends:

```
int RestClient::readResponse(String* response) {

  // an http request ends with a blank line
  boolean currentLineIsBlank = true;
  boolean httpBody = false;
  boolean inStatus = false;

  char statusCode[4];
  int i = 0;
  int code = 0;

  if(response == NULL){
    HTTP_DEBUG_PRINT("HTTP: NULL RESPONSE POINTER: \n");
  }else{
    HTTP_DEBUG_PRINT("HTTP: NON-NULL RESPONSE POINTER: \n");
  }

  HTTP_DEBUG_PRINT("HTTP: RESPONSE: \n");
  while (client.connected()) {     <================================== INFINITE LOOP HERE
    HTTP_DEBUG_PRINT(".");
    if (client.available()) {
      HTTP_DEBUG_PRINT(",");

      char c = client.read();
      HTTP_DEBUG_PRINT(c);

      if(c == ' ' && !inStatus){
        inStatus = true;
      }

      if(inStatus && i < 3 && c != ' '){
        statusCode[i] = c;
        i++;
      }
      if(i == 3){
        statusCode[i] = '\0';
        code = atoi(statusCode);
      }

      //only write response if its not null
      if(httpBody){
        if(response != NULL) response->concat(c);
      }
      if (c == '\n' && httpBody){
        HTTP_DEBUG_PRINT("HTTP: return readResponse2\n");
        return code;
      }
      if (c == '\n' && currentLineIsBlank) {
        httpBody = true;
      }
      if (c == '\n') {
        // you're starting a new lineu
        currentLineIsBlank = true;
      }
      else if (c != '\r') {
        // you've gotten a character on the current line
        currentLineIsBlank = false;
      }
    }
  }

  HTTP_DEBUG_PRINT("HTTP: return readResponse3\n");
  return code;
}
```

So this is a really HARD blocking fault and really disappointing :sweat:. The **TCPClient library is currently totally UNUSABLE with the Photon**:

- With an **available server**, the connect() works but the connected() doesn't work as expected (returning always 1 even then the socket is closed).

- With an **unavailable server**, the connect() doesn't work as expected (returning always 1) and the connected() makes the Photon PANIC after a previous "faulty" connect() call.

Tested with develop firmware versions 0.4.2 and 0.4.4-rc2, compiled with arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 in both cases.

[1]: https://github.com/llad/spark-restclient

anthonywebb, 2015-08-08:
Thanks for the hard work in running down this issue. Last week when we found our photons doing a panic/reset when making http calls we had to stop sending products to our customers.  Anxiously awaiting a fix.  Photons without TCPClient is a showstopper.

javier-rodas, 2015-08-08:
Thanks for the quick answer @anthonywebb

Knowing that you have detected it some time ago, and that you are working hard on solving this blocking issue really eases my mind :+1:

andyw-lala, 2015-08-10:
Validated both behaviours, and have small test cases to create both issues. Simply tries to connect to real/non-existent hosts and for the real host, there is either a netcat listener on the specified port or not.

I'm unsure why wiced_tcp_connect() would lie to us, unless it is an async call, and the current hal code does not register/use the callbacks that inform the caller of progress. Extending the timeout used in socket_connect() results in the wiced_tcp_connect() call taking a the appropriate amount of time, suggesting that the call is not asynchronous.

Interestingly enough, client.println() returns -3 on a connection that is doomed to failure, and that maps onto the WICED_PARTIAL_RESULTS error code.

Will update further as I delve deeper (modulo day-job.)

andyw-lala, 2015-08-10:
OK - it looks like #519 should fix the problems with connect(), but as my last add to that thread notes, it causes my development branch to SOS instead for any test cases that result in no connection.

Embarrassed that I stared at the damn function for so long and didn't see the mis-use of "wiced_result" vs "result", but that's what you get with variable names like that.

stevie67, 2015-08-12:
I tested with the new development branch and I can confirm that #519 fixes the connect issue for me. I can also confirm that the photon crashes afterwards. I debugged this a bit. What I found is that the call to socket_dispose in socket_close(...) crashes the photon. The socket->close() call still survives, but it never comes out of socket_dispose again.

monkbroc, 2015-08-12:
@stevie67 I reached all the same conclusions :) Good detective work. I wish I had the solution too.

stevie67, 2015-08-13:
Thanks @m-mcgowan, I came to the same conclusion that it is a bug in the WICED code. I put the wiced_tcp_delete_socket directly after the failed connect call and it immediately crashed in that call. No way I see this is outside the WICED code.

I tried a similar fix to yours (set the type to NONE, when the connect was not successful) over many hours now, with no apparent problem. It seems that if connect fails, the socket is actually disposed off automatically. Or something like this.

m-mcgowan, 2015-08-13:
Thanks for that information @stevie67 - good to know it's not leaking resources!

maghis, 2015-08-13:
guys, using it from the TCPClient, I don't have to worry about anything now, I can just call stop() no matter what, right?

stevie67, 2015-08-13:
@maghis this should not be a problem, it works for me.

anthonywebb, 2015-08-13:
Please advise once this has been posted to the cloud compilers used in particle dev and on the web.  I have to compile there for my product keys and product ID's to work properly for uploading the firmware into the dashboard.  I'm optimistic this will solve the http issues were have reported, excited to give it a try.

monkbroc, 2015-08-14:
I'm hopeful I've got a real fix to the crash for non-existent server. See #539.

m-mcgowan, 2015-08-17:
Gentlemen, I'm calling this one fixed. Thanks for the assistance @monkbroc!

closed 2015-08-17 by m-mcgowan

monkbroc, 2015-08-17:
Thanks for the credit @m-mcgowan!