particle-iot/device-os #491: Photon: Photon DFU Mode USB driver for Windows x64

XuGuohui, 2015-07-16:
Hi,

Have you released a usb driver for Windows x64 users when Photon in DFU Mode? I tried the driver located in "firmware-develop\misc\driver\windows", but failed.

Regards,
Guohui

m-mcgowan, 2015-07-16:
The driver should be compatible with 64-bit windows also. There is no executable code - just a signed .INF file.  When you say it failed, what do you mean? Please provide more information about the failure and how you tried to install the driver. 

Please follow the steps here for downloading and installing the driver: http://docs.particle.io/photon/connect/#connecting-your-device-using-windows

XuGuohui, 2015-07-16:
I followed the steps below:
1. Go to the Device Manager and double-click on your Particle device(Photon DFU Mode) under Other Devices.
2. Click Update Driver, and select Browse for driver software on your computer.
3. Navigate to the folder "firmware-develop\misc\driver\windows" and click on "Next".
4. Then a pop-up window prompt that Windows can not install Photon DFU Mode as it can not find a driver for this device.

kennethlimcp, 2015-07-16:
The driver in this repo is for Serial COM port.

For DFU in windows: https://community.particle.io/t/tutorial-installing-dfu-driver-on-windows-24-feb-2015/3518

XuGuohui, 2015-07-16:
@kennethlimcp ,
I'll try this and give feedback later.

XuGuohui, 2015-07-16:
@kennethlimcp , It works indeed! Thanks!

m-mcgowan, 2015-07-16:
Thanks @kennethlimcp you rock! @XuGuohui I'm pleased your issue was resolved.

closed 2015-07-16 by m-mcgowan