particle-iot/device-os #492: PWM outputs 500Hz on RX/TX pins and 625Hz on D0-D4/A4/A5/WKP pins

technobly, 2015-07-17:
Found that PWM outputs 500Hz on RX/TX pins and 625Hz on D0-D4/A4/A5/WKP pins. Should be 500Hz for all pins.

// The Photon should have 9 PWM pins: D0, D1, D2, D3, A4, A5, A7, RX and TX.
//
// EXAMPLE USAGE
// Outputs PWM on pwmPin
int pwmPin = D0;

void setup() {
  pinMode(pwmPin, OUTPUT);
  analogWrite(pwmPin, 64); // 25% PWM at 500Hz
}

satishgn, 2015-07-23:
Investigating #492

satishgn, 2015-07-24:
All PWM channels on Photon should now output 500Hz.

satishgn, 2015-07-24:
Fixed in branch: feature/photon-pwm-and-tone-fix (should later be merged with develop/latest before release)

closed 2015-07-24 by satishgn

m-mcgowan, 2015-08-17:
Just checked, this has been merged into develop. I deleted the branch now. (It's a good idea to create a PR from the branch - it's makes it easier to track when the changes have been merged.)

commit fd4dd30f (2015-07-24): Timer counter clock frequency is different on core/photon - Fixes #492