particle-iot/device-os #494: I2C doesn't allow 1.7Mhz

kennethlimcp, 2015-07-19:
When i set the I2C speed to 1.7Mhz, checking on `endTransmission` returns **4** and boils down to      the code `if(EVENT_TIMEOUT < (isr_safe_micros() - _micros)) return 4;` 

https://github.com/spark/firmware/blob/latest/hal/src/stm32f2xx/i2c_hal.c#L257

m-mcgowan, 2015-07-19:
Ah....! This makes sense. We switched over to `micros()` in response to https://github.com/spark/firmware/issues/460 and @maghis, author of that issue, assured me it worked.

At the time, I had forgotten that `micros()` wraps around after about a minute, so the subtraction trick to compute duration doesn't work.`micros()`. 

We have fixed `micros()` to use the 32-bit address space, but now this is interrupt based, so will not work with #460. I think the ideal fix is to use the `DWT->CYCCNT` counter to determine the timeout.

m-mcgowan, 2015-08-22:
@kennethlimcp  could you check this on 0.4.4 if you have time, and report if it works there?

kennethlimcp, 2015-09-02:
@m-mcgowan i did a quick test and `  Serial.print(Wire.endTransmission());` gave **3: end of address transmission timeout**

technobly, 2015-09-02:
@kennethlimcp does your device support 1.7MHz operation?  If not it won't NACK and this would cause the error you're seeing.  I've also not tested I2C at 1.7MHz... so it would be interesting to see if it's even clocking at that rate.

kennethlimcp, 2015-09-02:
I am using this I2C GPIO Expander MCP23008 - http://ee-classes.usc.edu/ee459/library/datasheets/MCP23008.pdf

technobly, 2015-09-02:
Lemme try the speed real quick... bbiab.

kennethlimcp, 2015-09-02:
For the record: https://community.particle.io/t/photon-i2c-still-unreliable/14948

technobly, 2015-09-02:
@kennethlimcp Ok so with this sample app, I was able to verify that 100000 and 400000 work fine, but when you get to 1700000, it's not quite right... mine is clocking out at 2MHz.  If I reduce the rate to 1500000 (1.5MHz) it is giving me 1.67MHz on my scope.

That said, it's very clear when looking at this on the scope that the capacitance on the bus is going to RULE THE SCHOOL at 1.7MHz.  I tested this on the Sparkfun Photon Weather Shield that I modified the pull up resistors on with 1k ohm pull ups, and it's still a very capacitive waveform (i.e., not a squarewave).  I'd say you'd need to add some 220 - 500 ohm pull up resistors to 3.3V (not 5V) to have a clean waveform.

```
void setup() {
    Wire.setSpeed(1500000);  // 1.67MHz
    Wire.begin();
}

void loop() {
    Wire.beginTransmission(0x55);
    Wire.write(0x55);
    Wire.write(0x55);
    Wire.endTransmission();
}
```

kennethlimcp, 2015-09-02:
@technobly cool i will test it out :) 0.4.4 is definitely giving more stability as we used to have other errors and now that we might narrow down to a pull-up value issue, having high speed support would be rad!

m-mcgowan, 2015-09-13:
Just bumping this, since it's not clear to me if we have a firmware bug or if this a manifestation of hardware issues.

peterqu, 2015-11-25:
A quick look at the STM32F2xx datasheet shows that the MCU supports I2C in Standard and Fast-Mode.
http://www.st.com/web/en/resource/technical/document/datasheet/CD00237391.pdf

Standard and Fast-Mode have maximum clock frequency of 100KHz and 400KHz respectively.  Page 46 of the doc linked below.
http://www.nxp.com/documents/user_manual/UM10204.pdf

So, I don't think it is a firmware bug.  The MCU simply does not support I2C clocked at higher than 400KHz.

technobly, 2015-11-25:
Thanks @peterqu 

From the STM32F205 Reference Manual:

>23.3.1 I2C functional description
>In addition to receiving and transmitting data, this interface converts it from serial to parallel format and vice versa. The interrupts are enabled or disabled by software. The interface is connected to the I2C bus by a data pin (SDA) and by a clock pin (SCL). It can be connected with a standard (up to 100 kHz) or fast (up to 400 kHz) I2C bus.

```
up to 400kHz
```

I think even though I was able to set the frequency higher, the slew rates of SCL and SDA will be too slow for it to work properly.  Going to close this out.

closed 2015-11-25 by technobly