particle-iot/device-os #496: develop: weird linker error when building system-part1

maghis, 2015-07-19:
I'm getting this when building latest develop:
```
`snprintf' referenced in section `.text.log_print_' of /tmp/ccycE68b.ltrans4.ltrans.o: defined in discarded section `.text' of ../../../build/target/system-part1/platform-6-m-lto/../../../modules/photon/system-part1/src/import_rt.o (symbol from plugin)
`vsnprintf' referenced in section `.text.log_print_' of /tmp/ccycE68b.ltrans4.ltrans.o: defined in discarded section `.text' of ../../../build/target/system-part1/platform-6-m-lto/../../../modules/photon/system-part1/src/import_rt.o (symbol from plugin)
collect2: error: ld returned 1 exit status
```

m-mcgowan, 2015-07-19:
What flags/defines are you passing to make?

maghis, 2015-07-19:
Running from modules/ (I have PARTICLE_DEVELOP=1 in my env):
```
make clean all PLATFORM=photon
```

m-mcgowan, 2015-07-19:
Looks to me like an issue with LTO - please add `COMPILE_LTO=n` to the commandline. I will push a change later that disables LTO for all modules apart from the bootloader, since adding LTO is exposing ARM GCC linker bugs which are causing us difficulties!

maghis, 2015-07-19:
@m-mcgowan confirmed! Thanks Matthew!

m-mcgowan, 2015-07-19:
My pleasure!

closed 2015-07-19 by m-mcgowan