particle-iot/device-os #501: Additional sleep modes and sleep API review

m-mcgowan, 2015-07-21:
As a recap, the current sleep API looks like this:

|                       |                     |
|-------------------|------------------|
| `System.sleep(60);` |  Put the wifi module to sleep for 60s. Application code continues running.|
| `System.sleep(D2, RISING, 60);` | Pause execution, using less power. Resume execution with the next statement after 60s, or when D2 transitions from LOW to HIGH, whichever is sooner.|
| `System.sleep(SLEEP_MODE_DEEP, 60);` | Stop execution, using much less power. Restart the system after 60 seconds or on a rising edge on the WKP pin (photon only).  The application restarts - memory is cleared, setup() is called.) |

- `System.sleep(60);` presently doesn't stop application code, only the WiFi module. This isn't obvious from looking at the API name. Perhaps this could be achieved using a clearer API such as `WiFi.sleep(60)`.

- There is no way to currently to pause execution and use less power for a period of time. (With pausing, the application would resume from where it left off.)

## Solution A

Make `System.sleep(60);` pause the the whole system (application and WiFi) rather than just WiFi. 

Pros:
- intuitive

Cons:
- existing code using this API will need to change

## Solution B

Add a new sleep mode, `System.sleep(SLEEP_MODE_PAUSE, 60)`

Pros:
- familiar, similar syntax to the existing `System.sleep(SLEEP_MODE_DEEP, 60)`
Cons:
- hard to remember, repeated use of word SLEEP which feels redundant.

## Solution C

Add a new API `System.pause(60)` that would pause the system for 60 seconds.

Pros:
- succinct
Cons:
- naming not related to sleep() so may be overlooked?

If we go for this short version of the API, we might consider also adding an alias for `System.sleep(SLEEP_MODE_DEEP,60);`

Possible candidates are:

- `System.stop(60);` - motivation: ties in with 'pause'
- `System.reset(60);` - motivation: the behavior is similar to performing a timed reset after the given timeout.

cc: @technobly, @jenesaisdiq, @satishgn

jenesaisdiq, 2015-07-21:
Great ideas captured here @m-mcgowan, would love to chat through them with you. When's good?

Initial thoughts: `System.` vs `App.` vs `WiFi.`... and whether `WiFi.` should become `Radio.` or similar so we don't have to change it again for Electron.

zsup, 2015-07-22:
Radio might not even be broad enough; what if we added Ethernet?

m-mcgowan, 2015-07-22:
Good points. Let's take the network discussion to https://github.com/spark/firmware/issues/502, so we can focus here on sleep modes.

m-mcgowan, 2015-09-13:
Regarding WiFi/Cellular/Radio/Ethernet...we can generalize all to `Network` so that the network subsystem can be shutdown to save power.

technobly, 2016-10-28:
Closing, superseded by https://github.com/spark/firmware/issues/1149

closed 2016-10-28 by technobly