particle-iot/device-os #503: Linker error: undefined reference to `_sbrk'

maghis, 2015-07-22:
Any idea what this is?

m-mcgowan, 2015-07-24:
It some function that is trying to allocate memory without going through `malloc`. If you can identify which function it s, we can add it to the system module so that this error disappears.

One way to do this is to add

```
void* _sbrk() { return 0; }
```

to your app, then use the `.lst` file produced to see which function is calling this.

maghis, 2015-07-24:
Uh, awesome, I didn't catch that. Is that because now prob malloc goes through dynalib?

m-mcgowan, 2015-07-25:
Yep. malloc is handled by the system modules, but some functions call _malloc_r_ which we can't redirect.

maghis, 2015-08-10:
Hey @m-mcgowan I'm still having a hard time w this.

I double checked and removed all the 'new' I had and switched to malloc, but I'm still getting the same error.
What else can I do?

maghis, 2015-08-10:
@m-mcgowan I found it.
It turns out that in my code I'm using mktime() somewhere and a bunch of Time functions.

The following are the 2 call stacks that lead to malloc_r:

Refresh_UnixTime_Cache
Convert_UnixTime_To_CalendarTime
localtime
localtime_r
_tzset_unlocked
_tzset_unlocked_r
malloc_r

mktime
_tzset_unlocked
_tzset_unlocked_r
malloc_r

What do you think is the best solution? Maybe expose Refresh_UnixTime_Cache and mktime via dynalib?

maghis, 2015-08-10:
Referencing
- https://github.com/spark/firmware/issues/526
- https://github.com/spark/firmware/issues/521

m-mcgowan, 2015-08-11:
The problem is indeed that some runtime functions are using malloc_r under the covers. _tzset_unlocked_r is one example. 

The fix is to move malloc_r to the system module. Sadly, this is more complex than would first appear since adding malloc_r to the dynalib is causing a duplicate symbol error.

wesner0019, 2015-08-11:
its strange i just flashed the latest development branch and my time functions worked. this development branch was the one with the changes found by andy

m-mcgowan, 2015-08-11:
Please double check your particular case - Andy's fixes were in a unrelated area. The problem here is a compiler issue with user firmware using Time functions when building modules/photon/user-part.

m-mcgowan, 2015-08-13:
@maghis  - could you please test the PR above and verify it fixes the issue. Thanks!

m-mcgowan, 2015-08-13:
Closing this now so we can more easily track oustanding issues. (I and several others have verified the fix - will reopen if issues are discovered by the OP.)

closed 2015-08-13 by m-mcgowan

maghis, 2015-08-14:
@m-mcgowan I'm testing now, should work, I'll reopen just in case it doesn't.

Thanks!