particle-iot/device-os #505: [Photon] "Verify device ownership" failed

XuGuohui, 2015-07-24:
Hi,

I have tried erasing the whole flash of Photon. I built the firmware-develop repository with PLATFORM=photon to obtain the bootloader firmware. I downloaded the system-part1, system-part2 and tinker firmware for photon in the release page of this repository. Then I used JFlash to program the bootloader to the Photon via SWD interface, it works in USB DFU mode with yellow LED binking. Then I updated the system firmware using dfu-util according to the wiki page:
https://github.com/spark/firmware/releases
The system firmware also updated succesfully since the blue LED begin blinking and the Photon appeared as a soft AP. Then I open the Particle App on my phone to "setup a Photon". Everything went smoothly until the step "Verify device ownership". 
Then I reset the Photon and it never appeared as a soft AP unless I kept holding the setup button for 10s after reset to erase the user profile.
Besides, I have deleted the Photon in my Particle account before I erasing the whole flash of Photon.
Can anyone tell me how to fix this? or is this a bug in the firmware or in App?

m-mcgowan, 2015-07-25:
Hello @XuGuohui  Please post your issue on our [forum](https://community.particle.io) where there are many more people available to help you, not least our wonderful team of Elites. Thanks!

XuGuohui, 2015-07-27:
@m-mcgowan Thank you for your suggestion. I'v created a topic in the particle community.
https://community.particle.io/t/photon-verify-device-ownership-failed/13838/1

closed 2015-07-27 by XuGuohui