particle-iot/device-os #507: 'stat: command not found' when building on Windows

jnm2, 2015-07-26:
When I follow the [Getting Started directions](https://github.com/spark/firmware/blob/develop/docs/gettingstarted.md), it succeeds up to the 'ARM GNU Create Flash Image' step. It creates `main.bin.pre_crc`, `main.bin.no_crc`, `main.elf` and `main.map`, as well as `.bin.pre_crc`, `.bin.no_crc`, `.elf` and `.map` files for my application name.

The build process worked for the Core firmware in 2014 when I followed [the instructions](https://github.com/spark/firmware/blob/master/README.md). Something has changed; the stat error is new.

Is there a workaround I can apply?

> [...]
> Invoking: ARM GNU Create Flash Image
> arm-none-eabi-objcopy -O binary ../build/target/main/platform-0/main.elf ../build/target/main/platform-0/main.bin.pre_crc
> if [ -s ../build/target/main/platform-0/main.bin.pre_crc ]; then \
>         head -c $((`stat --print %s ../build/target/main/platform-0/main.bin.pre_crc` - 38)) ../build/target/main/platform-0/main.bin.pre_crc > ../build/target/main/platform-0/main.bin.no_crc && \
>         tail -c 38 ../build/target/main/platform-0/main.bin.pre_crc > ../build/target/main/platform-0/main.bin.crc_block && \
>         test "0102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f20280078563412" = `xxd -p -c 500 ../build/target/main/platform-0/main.bin.crc_block` && \
>         ../build/bin/win32/sha256sum ../build/target/main/platform-0/main.bin.no_crc | cut -c 1-65 | xxd -r -p | dd bs=1 of=../build/target/main/platform-0/main.bin.pre_crc seek=$((`stat --print %s ../build/target/main/platform-0/main.bin.pre_crc` - 38)) conv=notrunc  && \
>         head -c $((`stat --print %s ../build/target/main/platform-0/main.bin.pre_crc` - 4)) ../build/target/main/platform-0/main.bin.pre_crc > ../build/target/main/platform-0/main.bin.no_crc && \
>          crc32 ../build/target/main/platform-0/main.bin.no_crc | cut -c 1-10 | xxd -r -p | dd bs=1 of=../build/target/main/platform-0/main.bin.pre_crc seek=$((`stat --print %s ../build/target/main/platform-0/main.bin.pre_crc` - 4)) conv=notrunc ;\
>         fi
> /usr/bin/sh: line 1: stat: command not found
> head: -38: invalid number of bytes
> make: *** [../build/target/main/platform-0/main.bin] Error 1

kennethlimcp, 2015-07-26:
Check with this: https://community.particle.io/t/toolchain-windows/11635

jnm2, 2015-08-01:
That works great, thank you. Would you consider adding a page in docs.particle.io and keep it up to date with the latest instructions on setting up and building for local cloud? It would be great to have one place to go to figure out all the details.

m-mcgowan, 2015-08-10:
Good idea @jnm2! At present, the docs for the build are in github, but we should definitely add a link to that form the main docs site so that everything is available/searchable from one place.

closed 2015-08-22 by m-mcgowan

jokkebk, 2016-02-21:
I encountered exactly the same problem. I was reluctant to use the "installed" provided in kennethlimcp's link, but the instructions directed me to add the mingw-developed-toolkit set of tools in MinGW Installation Manager. This installed about 56 new MinGW packages, but the stat.exe, xxd.exe and working version of test appeared afterwards.

So a short version of fix is to install those developer packages for MinGW. I think it _really_ sucks that MinGW web pages or installation manager don't have any search tool to figure out exactly which packages provided these couple of tools. :(

AmittSharma, 2016-11-12:
Hi all,
I followed the instruction as given above, but still I am getting this error:

/bin/sh: line 1: stat: command not found
/bin/sh: line 3: xxd: command not found
/bin/sh: line 3: test: 0102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f20280078563412: unary operator expected
make[1]: *** [../../../build/target/user-part/platform-103-m/user-part.bin] Error 2
make[1]: Leaving directory `E:/Electronics/Particle/Bluz/bluzDK-firmware/modules/bluz/user-part'
make: *** [modules/bluz/user-part] Error 2

Please help in this.
Thanks