particle-iot/device-os #508: time zone bug

tcrussell, 2015-07-27:
When the Time.zone() function is called, the output from Time.timestr() is updated correctly, but a call to Time.hour() reports the old time.  After a delay of 1 second, Time.hour is correct.  See code and output below:

void setup() {
    Serial.begin(115200);
    while(!Serial.available()) SPARK_WLAN_Loop();
    Serial.print(Time.timeStr());
    Serial.print("\rHour = ");
    Serial.println(Time.hour());
    Serial.println("Setting time zone to -4");
    Time.zone(-4);
    Serial.print(Time.timeStr());
    Serial.print("\rHour = ");
    Serial.println(Time.hour());
    Serial.println("Delay 1 second");
    delay(1000);
    Serial.print(Time.timeStr());
    Serial.print("\rHour = ");
    Serial.println(Time.hour());
}

void loop() {
    SPARK_WLAN_Loop();
}

OUTPUT:
Mon Jul 27 14:52:44 2015
Hour = 14
Setting time zone to -4
Mon Jul 27 10:52:44 2015
Hour = 14
Delay 1 second
Mon Jul 27 10:52:45 2015
Hour = 10

technobly, 2015-07-28:
Thanks for the example code @tcrussell, that will really help speed up reproducing this and tracking down the issue.  I'm going to add this to our internal backlog right now :)

technobly, 2015-07-28:
@tcrussell are you using a Photon or Core?  Also do you have the ability to try both?  The extra set of hands testing would be super helpful!

tcrussell, 2015-07-28:
I am using a Core.  I don't have a Photon to try it, but may order one soon.

Tom

commit b32b30bc (2015-08-13): calendar time updates immediately when timezone changed. Previously the cache was based on the unix time before adding the timezone. Now the cache is based on the unix time after adding the timezone, since this is what is actually reported in the various Time.hour(), Time.minute() etc. functions. fixes #508  FIRM-157
closed 2015-08-13 by m-mcgowan