particle-iot/device-os #509: Spark.process() blocks for extended periods of time, contrary to documentation

airbornemint, 2015-07-28:
I am trying to write an app for my Photon with a responsive touch interface. I was under the impression (based on the documentation for Spark.connect and Spark.process) that Spark.process was a lightweight call and could safely be called in what, for me, is the inner loop of my UI code. 

Unfortunately, I am discovering that Spark.process is, in some circumstances, taking a long time to return, which results in UI stutter, My UI refresh/sampling rate falls well below 10 Hz due to the delays in Spark.process.. This is undesirable.

I narrowed this down to the following test code:

    SYSTEM_MODE(SEMI_AUTOMATIC);
    
    void setup() {
        pinMode(D7, OUTPUT);
        digitalWrite(D7, LOW);
        Spark.connect();
    }
    
    void loop() {
        digitalWrite(D7, LOW);
        delay(100);
        digitalWrite(D7, HIGH);
    }

If you run this, you can eyeball the amount of time spent in Spark.process (implicitly invoked on each loop iteration) based on the brightness of the onboard LED on D7. What you will see is that while the Photon is going through its Wi-Fi and cloud handshake, ~100% of the time is spend outside of loop. It's only after the cloud handshake is complete that % of time spent outside of loop comes down to near zero. 

I had a quick look at the firmware code, and my best guess is that this is happening due to a number of blocking sends inside Spark_Communication_Loop.

(The behavior is the same in manual mode with an explicit call to Spark.process.)

My understanding from the documentation was that Spark.process was supposed to be a lightweight call and return within milliseconds (whereas Spark.connect is heavyweight and blocks for seconds at a time).

This is not what I actually observed, and my conclusion is that the only way I can get a good responsive touch UI out of a Photon is if I 

 a. Give up and drop a second MCU into this setup (so that it would run all the non-networking code and the Photon would be relegated to WiFi and Particle cloud duty) or 
 b. Turn off all networking code while the user is interacting with the device, and also turn off all interaction while there is network activity. 

I don't imagine that I have to explain why both of those solutions are problematic :-)

technobly, 2015-07-28:
@airbornemint thanks for the very detailed report.  Completely understand both cases not being ideal.  We'll have a look at this on a Photon and Core and see what if anything can be done about it with the current implementation of firmware.  I imagine once the true power of the FreeRTOS() is configured for multi-threading on the Photon, this will solve your issue.  We've been working on implementing this for the near future ;-)

airbornemint, 2015-07-28:
Oh, if that's where you're at, then let me ask a followup so I can prioritize my time better: is it your goal for the 2015 Photon release to have the user code and the network/cloud code scheduled deterministically? Because if the answer is yes, then I am just going to plow ahead with other parts of my project and wait for the scheduling issues to be resolved by the particle.io team, rather than spinning my wheels writing code in my project to work around the current scheduling problems. :-)

bkobkobko, 2015-07-28:
See this thread in the community for discussion:

https://community.particle.io/t/spark-process-blocks-for-extended-periods-of-time-contrary-to-documentation/13862

m-mcgowan, 2015-08-10:
@airbornemint - we will be adding multithreading this quarter to the Photon so that application code and the networking code are scheduled deterministically. (FreeRTOS uses a 1ms timeslice, so that should be the maximum length of time your code is blocked for.)  So if you can hop over this issue for now and focus on other parts, you'll find the issue is resolved with a future firmware update on the Photon!

airbornemint, 2015-08-10:
Much thanks. I'll muddle through. :-)

m-mcgowan, 2015-09-30:
Multithreading beta will be released in 0.4.6, which addresses this so closing out.

closed 2015-09-30 by m-mcgowan