particle-iot/device-os #510: The latest branch README seems to have broken "flash" instructions

joltup, 2015-07-29:
The current flash instructions talk about the "make" command?

____________________________________________________________
Now, from the main/ folder in your firmware repository and use the following command to transfer the .bin file into the device.
````
make program-dfu
````
  Upon successful transfer, the device will automatically reset and start the running the program.
__________________________________________________________________________

**Whereas it on master it makes more sense**:

__________________________________________________________________________
Now, navigate to the build folder in your firmware repository and use the following command to transfer the .bin file into the Core.
````
dfu-util -d 1d50:607f -a 0 -s 0x08005000:leave -D core-firmware.bin
````
For example, this is how my terminal looks like:
````
D:\Spark\firmware\build [master]> dfu-util -d 1d50:607f -a 0 -s 0x08005000:leave -D core-firmware.bin
````
  Upon successful transfer, the Core will automatically reset and start the running the program.
__________________________________________________________________________

From what I built, I feel like it should be:
````dfu-util -d 2b04:d006 -a 0 -s 0x08005000:leave -D user-part.bin````

kennethlimcp, 2015-07-30:
Using `program-dfu` will directly execute the `dfu-util xxxx` command without having to switch directories and copy/paste the lengthy command.

technobly, 2015-08-10:
@joltup when compiling the user-part (aka application firmware) locally I would recommend compiling from the main directory as:
`make clean all PLATFORM=photon -s program-dfu`

This will clean the build directory, and make all targets as the photon platform, without verbose output, and then finally run the dfu-util command required to transfer the binaries to your device.

The Core simply didn't have this `program-dfu` macro available, so you had to type it in manually.

Hopefully this makes sense, but if you still have questions please post them on the forums here where you'll get a much faster response: https://community.particle.io

closed 2015-08-10 by technobly