particle-iot/device-os #511: WiFi connect() and listen() methods not functioning as documented

LeoWW, 2015-07-30:
Here's the situation:

Build environment
1. I am compiling the code on photons
2. I use the "develop" branch of the system firmware
3. I have the photons flashed with the 0.4.3 update

Observations:

WiFi.connect()

If the Photon has credentials, it will enter connecting mode (blinking green) and attempt to connect to WiFi and block user code. If the Photon doesn't connect, it will attempt to connect for approximately 30 to 40 seconds (duration varies each time) and then exits to user code.
Sometimes, the Photon will attempt to connect but immediately exit out of connecting mode (blinking green) and start running user code.
After exiting out of connecting mode (blinking green), the photon will continue blinking green while running user code until a different wifi function is called.

WiFi.listen()

Blocks user code indefinitely (already noted feature)
When called, will sometimes immediately enter listening mode (blinking blue) and may sometimes take up to 20 seconds to enter listening mode i.e. if in connecting mode (blinking green), will take up to 20 seconds before the photon begins blinking blue and broadcasting its wifi signal.

m-mcgowan, 2015-08-10:
Thanks for reporting! Although I'm not sure I see the issue clearly. Could you explain how the behavior you're seeing is different from what you're expecting?

If the device is busy connecting to WiFi, then your application code (WiFi.listen()) isn't running - the LED flashes green, which might be why WiFi.listen() takes a long time to connect.

LeoWW, 2015-08-10:
Hey mdma, welcome back! 

The issue is (1) WiFi.connect() does not always perform the same behavior and (2) blocks user code for an unknown amount of time.

On some startups of the photon (I would say about 10% of the time), when WiFi.connect() is called it will immediately run user code and not attempt to connect to wifi. This is disconcerting because it usually attempts for 30-40 seconds so it seems to me that the function call is not very robust. 

Example: If I have code that calls WiFi.connect() and then WiFi.listen() immediately afterwards and the photon has wifi credentials but there is no available wifi network, then the photon will sometimes immediately run through WiFi.connect() (blinking green for < 1 second) and run WiFi.listen() (start blinking blue for listening mode). In this situation it seems that the photon doesn't actually try to connect to a wifi network. Expected behavior is WiFi.connect() will block user code for ~30-40 seconds and then run WiFi.listen() to enter listening mode. 

However, the desired behavior is for WiFi.connect() to not block user code (i.e. like running a system event, which is what you implemented for WiFi.listen() :smile:), or, at least, that we know and can control how long WiFi.connect() will run (i.e. if we want WiFi.connect() to run for 30 seconds, we can tell it to run for 30 seconds and it will block user code for exactly 30 seconds each time). If this could be implemented that would be awesome! :smile:

Here is some example code for testing this behavior 
```c++
#include "application.h"

SYSTEM_MODE(MANUAL);

long unsigned retrymillis = 0;
uint32_t counter = 90;
uint32_t stopTime = counter*1000;
bool lightonce = false;

//This is a system event that acts as an ISR. This function allows for us to exit
//out of listening mode.
void listen(system_event_t event, uint32_t param, void* pointer)
{
    if(millis() > (retrymillis + 1000))
    {
    	if(!(lightonce))
    	{
    		Serial1.begin(38400);
    		Serial1.print("$0090L1399990000.");
    		lightonce = true;
    		Serial.println("Light on");
    	}

        Serial.print("Time left: ");
        counter--;
        Serial.println(counter);
        retrymillis = millis();
    }

    if(event == wifi_listen_update)
    {
        if(param > stopTime)
        {
            WiFi.listen(false);
        }
    }
}

/* This function is called once at start up ----------------------------------*/
void setup()
{
	WiFi.on();
	Serial.println("WiFi Turned On");

	while(!(WiFi.hasCredentials()))
	{
		WiFi.setCredentials("GenericUser","GenericPass");
		delay(100);
		Serial.println("Credentials Set");
	}

	if(!(WiFi.ready()))
	{
		Serial.println("Attempting to Connect");	
		WiFi.connect();
		Serial.println("After Connect Call");
	}

	Serial.println("NOW GOING INTO LISTENING MODE");
	System.on(wifi_listen,listen);
	WiFi.listen();
	if(WiFi.listening())
		Serial.println("In Listening Mode");
	Serial.println("This line runs after WiFi.listen() because there is no delay() call between this line and WiFi.listen()");
	Serial.println("In void loop() before delay(1000)");	
	delay(1000);
}

/* This function loops forever --------------------------------------------*/
void loop()
{
	if(!(Spark.connected()))
		Spark.connect();
	else
		Spark.process();

	delay(1000);
	Serial.println("after delay(1000)");
	//listening mode blocks user code and both the serial and serial1 channel. 
}
```

m-mcgowan, 2015-08-13:
Taking the last point first, the docs are not 100% correct - the device will eventually give up in around 30 seconds and retry after another 30 seconds.

m-mcgowan, 2015-08-13:
@LeoWW  - when you get chance, would be great if you could verify the change above improves the situation!

LeoWW, 2015-08-14:
Thanks for the quick response! Will try the latest fixes tomorrow and comment here with the results! 

Just a heads up, I did notice today that pressing the setup button for 3 seconds to enter listening mode still causes the weird LED behavior on the photon (i.e. LED turns solid green for a few seconds before blinking blue and entering listening mode. 

Fortunately enough, calling WiFi.listen() from user code immediately causes the photon to enter listening mode. I definitely have noticed before that the photon re-enters listening mode if the photon failed to connect to wifi. Can't wait to test the commit you made that fixes that! :smile: 

Good luck with your other bug fixes, I see you've been hard at work :+1:

LeoWW, 2015-08-14:
Just tested possible fix for #511 (this issue) -- If the photon receives credentials and doesn't connect to wifi, the photon will re-enter listening mode instead of running user code -- is there a way to call wifi.listen(false) if the photon is not able to connect to wifi after receiving credentials?

LeoWW, 2015-08-14:
I've also noticed that WiFi.connect() attempts to connect for 7 seconds vs. 30 seconds before the changes

m-mcgowan, 2015-08-17:
Howdy, yes I there's a way to get what you're after:

- add a handler for the system event `wifi_listen_end` - this signifies that listening is about to finish.
- call `WiFi.listen(false)` which will force listening to end manually and not revert back to listening mode if the connection isn't successful.

I hope that fixes your issue!

m-mcgowan, 2015-08-21:
The steps above are the same as the workaround code pasted in #558.  Does that resolve this issue?

m-mcgowan, 2015-08-28:
Closing out - assumed resolved.

closed 2015-08-28 by m-mcgowan