particle-iot/device-os #521: Develop branch Time class calling mallocr instead of malloc

wesner0019, 2015-08-07:
[quote="peekay123, post:17, topic:14113, full:true"]
Folks, it turns out that sbrk() is called by functions in the Time class which are calling mallocr instead of malloc.  When @mdma gets back, he should be able to fix this quickly.  This only affects the develop branch.
[/quote]

bprobbins, 2015-08-07:
Just git pulled today on the latest branch and found that the sbrk bug now afflicting the latest branch today as well

technobly, 2015-08-10:
FWIW: I tried replacing localtime() with localtime_r() and it still requires _sbrk(), but it might be worth considering using the reentrant safe version of localtime anyway.

m-mcgowan, 2015-08-10:
afaik, localtime() calls localtime_r(), and that's true for all standard and re-entrant functions, the standard function calls the re-entrant one using a global context.

m-mcgowan, 2015-08-11:
Duplicate of #503.

closed 2015-08-11 by m-mcgowan