particle-iot/device-os #522: HAL pinmap #define's NONE causing conflict

bkobkobko, 2015-08-08:
A user reports that his enums don't work due to name collision with a `#define NONE ...` in the pinmap file.  This should be changed to better name or better scope to control the conflict.

https://community.particle.io/t/hal-layer-defines-clash-with-enums-in-user-code-photon/14210

technobly, 2015-08-09:
Interesting, this has existed since the Core:
https://github.com/spark/firmware/blob/master/inc/spark_wiring.h#L65

And is carried on in the Photon:
https://github.com/spark/firmware/blob/develop/hal/src/stm32f2xx/pinmap_hal.c#L31
https://github.com/spark/firmware/blob/develop/hal/src/stm32f2xx/pinmap_impl.h#L55

Agreed that "NONE" is probably too generic of a name.  Once upon a time we had used LED_PIN as well ;-)  I'll add this to a task so we don't forget about it.

m-mcgowan, 2015-08-10:
ah...defines...like marmite you either love or hate it! :-) I've tried to fix this in the past by making it a constant, but that doesn't play well with pure-C files (since const int X=5 isn't considered a compile time constant in C.)

A short term fix is to change the scope of the define, or rename it to something more unique.

m-mcgowan, 2015-08-13:
Looking into this, the constants were renamed in commit https://github.com/spark/firmware/commit/84ef39351b1874bd8e6700bfb76fded440be39a6.

We are just waiting for 0.4.4 to be deployed to the compile farm for these changes to take affect in the Web IDE.

closed 2015-08-13 by m-mcgowan