particle-iot/device-os #523: build (web) seems to ignore #include for libraries attached to sketch via web UI

plocher, 2015-08-08:
After attaching LiquidCrystal_I2C_Spark to my cp-test sketch*, the web UI added the expected

```C
#include "LiquidCrystal_I2C_Spark/LiquidCrystal_I2C_Spark.h"
```

The compile then fails because of the (already reported) missing CLOCK_SPEED_100KHZ define, and the suggestion to include the i2c_hal header file doesn't fix things for me...

In order to continue my porting effort (yeah, I've got bugs of my own to track down, too!), I commented out the #include and the lcd variables/uses related to the I2C LCD backpack.

```C
//#include "LiquidCrystal_I2C_Spark/LiquidCrystal_I2C_Spark.h"
#ifdef HASLCD
//LiquidCrystal_I2C lcd(0x20 + 7, 16, 2);  // Set the LCD I2C address
#endif
```
etc

After saving and verifying my sketch, I was surprised that it still fails - somehow the unwanted and commented out library still seems to be used!

```C
LiquidCrystal_I2C_Spark/LiquidCrystal_I2C_Spark.cpp: In member function 'void LiquidCrystal_I2C::init_priv()':
This looks like an error in LiquidCrystal_I2C_Spark library. Would you like to create an issue on GitHub to let the author know?
CREATE ISSUE
LiquidCrystal_I2C_Spark/LiquidCrystal_I2C_Spark.cpp:48:23: error: 'CLOCK_SPEED_100KHZ' was not declared in this scope
         Wire.setSpeed(CLOCK_SPEED_100KHZ);
                       ^

make[1]: *** [../build/target/user/platform-6LiquidCrystal_I2C_Spark/LiquidCrystal_I2C_Spark.o] Error 1
```

Is this a temp file cache error, the way libraries "work" in build.io, or worse, a facepalm moment for me :sweat_smile:  ?

----
*Happy to share the sketch ID if needed...

dmiddlecamp, 2015-08-08:
Hi @plocher,

  Good question!  When the library is included with your project in the Build IDE, the files are included during the build, even if you don't have an ```#include``` statement in your code.  I think that's what you're hitting?

Thanks!
David

plocher, 2015-08-09:
That seems to be correct, but when I create a simple test sketch on build.io (based off of elapsedMillis example code), with NO user added .h and .cpp files, commenting out the include causes compile errors.

This seems to imply that the #include IS including the header file (and commenting it out does not), but the LINK phase still tries to compile and use the associated library cpp code.  This is borne out by the inability to #include arbitrary public libraries used by other sketches in build.io without first using the library manager tab to "add" them to this sketch.

i.e., the build.io web's add library sets some state outside of the sketch source tree that keeps track of the libraries used independently of the state of the actual #include in the source; in the Arduino world, this state is derived from the source code by (effectively) doing a grep "^#include" pass at the beginning of the compile step.

dmiddlecamp, 2015-08-09:
Sure, that makes sense.  Thanks for reporting that distinction, it sounds like the makefile for the user app isn't quite properly excluding files that aren't strictly included.

m-mcgowan, 2015-08-13:
Adding `#include "i2c_hal.h"` will fix references to `CLOCK_SPEED_100KHZ` in the same file that you add the include to, but not to references from within in the I2C library itself. 

These issues have been fixed and are pending rollout of 0.4.4 to the cloud.

m-mcgowan, 2015-08-13:
@plocher  If you're satisfied with this resolution please close the issue, or let us know how we can help further. Thanks!

plocher, 2015-08-16:
The CLOCK SPEED issue was simply the penultimate itch that uncovered this issue, and is not really relevant to this post :-)

This issue is:

The act of telling the web UI to use a library seems to be the only way to inform the compile/link phase that a library is used, and that act is independent of whether or not the user #includes the header file for the library in their ino file.

Proof points:
*  simply adding a #include for a known library without interacting with the UI fails (linker doesn't find it...), 
*  after UI-adding-a-library, commenting out the #include doesn't eliminate the library from the compile/link phase, and
*  the linker/makefile/<whatever> logic interacts in some way with the web "flash" functionality, doing a "processing-like scan of the ino file for includes", but that functionality doesn't seem to be tied into either the compile or link phases ( see https://community.particle.io/t/build-io-flash-logic-warns-about-valid-but-moved-includes/14237/7 )

If you are going to implement an Arduino-like Processing-based grep-for-includes, you should not need to have any magic makefile state created by using the web UI's "add library" functionality - the presence or absence of the #include should be sufficient for both the compile and link phases.

The Web UI's "add library" should simply and only add a #include line to the ino file.

The build/flash web UI should look for #includes and create the compile and linker actions based on what it finds.  I'd suggest working with T.igoe  and PaulS in the Arduino/Teensy world to figure out the best ways to do this, as neither the Particle or Arduino mechanisms are terribly well done :-(

If I were god (:smile:), I'd put resources into fixing the persistent "Adding a library duplicates all your non-ino tabs" bug first, then this one as a lower priority annoyance...

Putting things into perspective, though, I'd put "update Particle to the latest Arduino/Wiring infrastructure with your HAL layer so I can use the entire Arduino ecosystem" ahead of this :-)

m-mcgowan, 2015-08-16:
Seems like you're expecting things to work differently from how they presently work. I think the key issue is that we don't do arduino-like grep processing, and IMHO that's a good thing - inferring dependencies by scanning headers is potentially very unreliable, especially since arduino library headers aren't in separate namespaces, so the possibility of name-clashes is real. Additionally scanning headers doesn't tell you which version of a library to use.

plocher, 2015-08-17:
> we don't do arduino-like grep processing

But you do, and that's the rub ( see
https://community.particle.io/t/build-io-flash-logic-warns-about-valid-but-moved-includes/14237/7
 )

I could see doing it another way, but it looks like you do it both ways at
different times :-(

Y'all have set the expectation that Particle is Arduino-like, and now we're
quibbling about how much "-like" there should be.  IMO, the status quo of
being partially compatible and partly incompatible is the worst place to be
- I call this "Arduino--++" - where it should be "Arduino++".

> Seems like you're expecting things to work differently from how they
presently work.

Yup - that's the nature of bugs :-)

The Particle environment isn't like Unix C on a command line with
Makefiles, it isn't like Arduino's native IDE, it isn't like Arduino's Web
IDE, and yet, it tries to be like them all.  And is failing.  In being
arbitrarily different from all of them, and yet not significantly better,
the differences become a liability.  Worse, because of the existing bugs
and fork-based technical debt, the Particle IDE environment is no longer as
good as the Arduino one - in the last year they have taken a stab or three
at the problems you point out, and gotten better - though I agree that the
Processing-style include/lib handling is an abomination and needs to
drastically improve  :-)

As I said, this is NOT a show stopper bug.  Don't waste more time on it -
go fix the duplicate tabs problem.

  -John

m-mcgowan, 2015-08-17:
> As I said, this is NOT a show stopper bug.  Don't waste more time on it -
go fix the duplicate tabs problem.

This is the firmware repo. For IDE bugs, please file an issue with the IDE repo.

closed 2015-08-23 by m-mcgowan