particle-iot/device-os #525: [Core]

WebDust21, 2015-08-09:
When using the LATEST build on a Core, analogRead() does not return the full 0-4095 range as per the spec.  With A(x) pin shorted to either GND or 3V3*, I get the following results:
ADC_SampleTime_1Cycles5: 0v -> 4.  3.3V -> 4091
ADC_SampleTime_7Cycles5: 0v -> 2.  3.3v -> 4092  (this the default setting)
ADC_SampleTime_28Cycles5: 3.3v -> 3887 (didn't note 0v)
ADC_SampleTime_239Cycles5: 3.3v -> 3887.

I can provide further details and samples if necessary.

WebDust21, 2015-08-09:
Reading the STM32F103 reference manual, I think I understand what is going on, causing this phenomena.  
As in:
````
11.9.4  Slow interleaved mode 
This mode can be started only on a regular channel group (only one channel). The source of 
external trigger comes from regular channel mux of ADC1. After external trigger occurs: 
• ADC2 starts immediately and
• ADC1 starts after a delay of 14 ADC clock cycles.
• ADC2 starts after a second delay of 14 ADC cycles, and so on.
Note: The maximum sampling time allowed is <14 ADCCLK cycles to avoid an overlap with the 
next conversion.
````

That last line is the kicker.  The default Particle setting of ADC_SampleTime_7Cycles5 is already beyond the specified maximum of 14 ADCCLK cycles for slow interleaved mode, with a total of 20 ADCCLK cycles per conversion.  This explains why only "1Cycles5" and "7Cycles5" give closest to full range.  Needless to say, when increasing the sample time, this only gets worse.  Basically, as "analogRead" uses slow-interleaved mode, setADCSampleTime is useless.

The end effect is that instead of increasing the impedance with longer sample times, input impedance is being LOWERED by having both A/D modules sampling at the same time.  I can prove that by taking a 10K resistor from an A(x) input to 5v--the reading goes straight to 4095, and with a meter, the pin is at 3.6v.

technobly, 2015-08-10:
Exactly right @WebDust21.  Per the community discussion, I look forward to testing your PR :)

m-mcgowan, 2015-08-17:
We have this PR already - will that work? https://github.com/spark/firmware/pull/258

WebDust21, 2015-08-17:
@m-mcgowan: The PR you referenced also removes the 10-step DMA-read averaging from analogRead(), replacing it with a single ADC conversion result.  

Will that PR work?  It depends on whether you want to maintain the stabilizing effect of a 10-step averaged ADC read, or not.

closed 2020-02-28 by avtolstoy