particle-iot/device-os #526: [Photon] develop firmware 0.4.4-rc2: TIME library causing link error (undefined reference to `_sbrk')

javier-rodas, 2015-08-09:
Tested with the develop firmware **0.4.4-rc2** and arm-none-eabi-gcc 4.9.3 **20150529**.

**NOT** happening with the firmware **0.4.3** and arm-none-eabi-gcc 4.9.3 **20150306**.

Test code:

```
#include "Application.h"

void setup() {
	int hour = Time.hour();
}

void loop() {
}
```

Link error:

```
$ make clean all PLATFORM=photon APP=test
...
Building target: ../../../build/target/user-part/platform-6-m/test.elf
Invoking: ARM GCC C++ Linker
mkdir -p ../../../build/target/user-part/platform-6-m/
arm-none-eabi-g++ -DSTM32_DEVICE -DSTM32F2XX -DPLATFORM_THREADING=1 -DPLATFORM_ID=6 -DPLATFORM_NAME=photon -DUSBD_VID_SPARK=0x2B04 -DUSBD_PID_DFU=0xD006 -DUSBD_PID_CDC=0xC006 -DINCLUDE_PLATFORM=1 -fno-builtin -DUSE_STDPERIPH_DRIVER -DDFU_BUILD_ENABLE -fno-builtin-malloc -fno-builtin-free -fno-builtin-realloc -DUSER_FIRMWARE_IMAGE_SIZE=0x20000 -DUSER_FIRMWARE_IMAGE_LOCATION=0x80A0000 -DMODULAR_FIRMWARE=1 -DMODULE_VERSION=3 -DMODULE_FUNCTION=5 -DMODULE_INDEX=1 -DMODULE_DEPENDENCY=4,2,4 -DRELEASE_BUILD -Werror -I./inc -I../../../user/inc -I../../../dynalib/inc -I../../../services/inc -I../../../hal/inc -I../../../hal/shared -I../../../hal/src/photon -I../../../hal/src/stm32f2xx -I../../../hal/src/photon/api -I../../../system/inc -I../../../rt-dynalib/inc -I../../../wiring/inc -I../../../modules/photon/system-part1/inc -I../../../platform/shared -I../../../platform/MCU/STM32F2xx/CMSIS/Include -I../../../platform/MCU/STM32F2xx/CMSIS/Device/ST/Include -I../../../platform/MCU/STM32F2xx/SPARK_Firmware_Driver/inc -I../../../platform/MCU/STM32F2xx/STM32_StdPeriph_Driver/inc -I../../../platform/MCU/STM32F2xx/STM32_USB_Device_Driver/inc -I../../../platform/MCU/STM32F2xx/STM32_USB_Host_Driver/inc -I../../../platform/MCU/STM32F2xx/STM32_USB_OTG_Driver/inc -I. -MD -MP -MF ../../../build/target/user-part/platform-6-m/test.elf.d -ffunction-sections -fdata-sections -Wall -Wno-switch -Wno-error=deprecated-declarations -fmessage-length=0 -fno-strict-aliasing -DSPARK=1 -DSTART_DFU_FLASHER_SERIAL_SPEED=14400 -DSTART_YMODEM_FLASHER_SERIAL_SPEED=28800 -g3 -gdwarf-2 -Os -mcpu=cortex-m3 -mthumb ../../../build/target/user-part/platform-6-m/src/module_info.o ../../../build/target/user-part/platform-6-m/src/user_export.o ../../../build/target/user-part/platform-6-m/src/user_module.o ../../../build/target/user-part/platform-6-m/src/newlib_stubs.o  --output ../../../build/target/user-part/platform-6-m/test.elf -Wl,--whole-archive ../../../hal/src/photon/lib/STM32F2xx_Peripheral_Libraries.a -Wl,--no-whole-archive --specs=nano.specs -lnosys -L../../../modules/photon/system-part2 -L../../../modules/photon/system-part1 -L. -T./linker.ld -Wl,--defsym,USER_FIRMWARE_IMAGE_SIZE=0x20000 -Wl,--defsym,USER_FIRMWARE_IMAGE_LOCATION=0x80A0000 -Wl,-Map,../../../build/target/user-part/platform-6-m/test.map  -L../../../build/target/user/platform-6-m/applications/test/ -L../../../build/target/services-dynalib/arm/ -L../../../build/target/hal-dynalib/platform-6-m/ -L../../../build/target/system-dynalib/platform-6-m/ -L../../../build/target/rt-dynalib/platform-6-m/ -L../../../build/target/wiring/platform-6-m/ -L../../../build/target/communication-dynalib/platform-6-m/ -L../../../build/target/platform/platform-6-m/ -L../../../hal/src/photon/lib/ -Wl,--whole-archive -luser -lhal-dynalib -lservices-dynalib -lsystem-dynalib -lrt-dynalib -lwiring -lcommunication-dynalib -lplatform -Wl,--no-whole-archive -L../../../build/arm/linker -nostartfiles -Xlinker --gc-sections
/usr/local/Cellar/gcc-arm-none-eabi-49/20150609/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/lib/armv7-m/libg_nano.a(lib_a-sbrkr.o): In function `_sbrk_r':
sbrkr.c:(.text._sbrk_r+0xc): undefined reference to `_sbrk'
collect2: error: ld returned 1 exit status
make[1]: *** [../../../build/target/user-part/platform-6-m/test.elf] Error 1
make: *** [modules/photon/user-part] Error 2

$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 224288]
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

BracWebb, 2015-08-09:
I believe this issue, commented on here: https://community.particle.io/t/photon-listening-mode/12870/33?u=bracwebb

is already captured in issue #521 
https://github.com/spark/firmware/issues/521

m-mcgowan, 2015-08-11:
Thanks for the report. Closing since this is a duplicate of #521.

closed 2015-08-11 by m-mcgowan