particle-iot/device-os #528: Photon can function as a HID device

towynlin, 2015-08-11:
This has been discussed for a long time in the community, and it has been in progress as a low priority story for a long time. Adding this issue mostly as a place for discussion. If you want this, please add a :+1: in the comments.

wgbartley, 2015-08-11:
:+1: :heart:

ScruffR, 2015-08-11:
:+1:
Especially cool would be the option to have a composite USB HID device (keyboard + mouse + serial + ...) 
A model implementation for this would be Teensy, where you can have each combination of HIDs with or without USB serial.

WebDust21, 2015-08-11:
Oh, great...as if the Windows "serial ballpoint mouse" wasn't a big enough problem, now you can do it on purpose??  I give a thumbs up anyway :-)  :+1:

technobly, 2015-08-11:
I'm going to add one for @rpiswag on the forums :+1:

harrisonhjones, 2015-10-06:
:+1: Would be nice to have a macro like: 

HID_SET_TYPE(NONE)
HID_SET_TYPE(HID_MOUSE || HID_KEYBOARD) For mouse and keyboard
HID_SET_TYPE(HID_MOUSE || HID_SERIAL) for mouse and serial
... etc ...

ScruffR, 2015-10-06:
:+1: :+1: :+1: :+1: :+1: :+1: :+1:  for the macro idea

rfresh737, 2015-12-07:
Mouse.moveTo() absolute positioning would be a very welcome feature.

hramadan, 2015-12-19:
Ten thumbs up from me! Is it possible to get an ETA for a firmware release? Am sure that I am not the only one hanging out for this feature......

greg5678, 2015-12-31:
:+1: :+1: :+1: :+1: :+1: :+1: :+1: for the macro idea! Please!

jtzemp, 2015-12-31:
Is it possible just to have a small, lightweight shim in the firmware and to have the bulk of the implementation in an external library? I'm still pretty noobish when it comes to firmware-land, but I do have interest in doing some HID-related stuff. I'm mostly just thinking about taking up space on the device for a `Mouse` and `Keyboard` library that would only get used in a fraction of the cases where people are using Photons (and Electrons, I suppose since it uses the same MCU).

ScruffR, 2015-12-31:
I'd be happy if it was an extra library too, but AFAIK things like this won't be linked into the BIN if not used anyway.

hramadan, 2015-12-31:
When enabling multiple USB device types, it would be nice to only enable the ones you want at *runtime* so that only the devices you are going to use end up in the PC's device list. This allows the USB device types to be selected by the user via config or command.

m-mcgowan, 2016-01-01:
Totally JT. Any significant functionality is best pushed out to a library if the technical constraints allow that. (I did the same with FreeRTOS on the core - a few shims in firmware and the bulk of the freertos code available as a library.)

bymaximus, 2016-01-25:
Forecast? :+1:

naikrovek, 2016-02-12:
:+1: this would be very handy for me.

NGenetzky, 2016-02-16:
I think this would be very helpful.

closed 2016-06-15 by technobly

jango-fx, 2016-09-23:
so is this working now?
is there any documentation of it??

ScruffR, 2016-09-23:
Yes it is, but only for the STM32F2 devices Photon/P1/Electron and not for the STM32F1 Core

jango-fx, 2016-09-23:
and how?

hramadan, 2016-09-23:
@jango-fx, the following should help:

https://community.particle.io/t/solved-0-6-0-rc2-keyboard-press-how-to/25704

jango-fx, 2016-09-23:
all I get is
`error: 'Keyboard' was not declared in this scope`

hramadan, 2016-09-23:
@jango-fx, how are you compiling, with the Web IDE?

#include "application.h"

ScruffR, 2016-09-23:
I guess this discussion should not be lead in the comments section of the issue.
@jango-fx, you can open a new thread at https://community.particle.io or post a reply on the topic linked above.

jango-fx, 2016-09-23:
hm, didn't find that thread during my research... switched to the forum. thx so far.

PR #902 (2016-06-15): Composite USB device driver implementation for STM32F2 (Photon/Electron/P1)
By default Photon/Electron and P1 now present themselves as a composite USB device with two USB Serials, `Serial` being the first and a new `USBSerial1` being the second one. This should work out of the box on Linux and Mac, Windows users will have to reinstall serial drivers. (This PR includes UNSIGNED drivers in `misc/driver/windows`, see 9b62479d6acd85be487356b8c50c75fcf260dfc1).

This PR also implements HID support (closes #528). Calling Mouse.begin() or Keyboard.begin() will register HID class driver in Composite  [... cut]