particle-iot/device-os #536: TCPClient stop() does not work on first connection

nylen, 2015-08-14:
I wrote a simple HTTP server for the Photon.  When a client connects, I write a basic HTTP response in a few lines, then call `client.flush()` and `client.stop()`.

This works well, except for one thing:  my code does not disconnect the very first client to connect to the Photon after it boots up, so the connection hangs.  Every connection after the first one works fine.

Here's my full code:

```
int led1 = D6; // connected to solid-state relay
int led2 = D7; // the little blue LED built into the board
int buttonLow = D2;
int buttonInput = D1;
int buttonHigh = D0;

int ledState = HIGH;

TCPServer server = TCPServer(80);
TCPClient client;
char addr[16];

const int MAX_HTTP_LENGTH = 255;
char httpBuffer[MAX_HTTP_LENGTH + 1];
int httpLength;

void setup() {
    pinMode(buttonLow, OUTPUT);
    digitalWrite(buttonLow, LOW);
    
    pinMode(buttonInput, INPUT);
    
    // Writing an INPUT mode pin HIGH does not seem to work here,
    // like I hear it does on Arduino
    // so we're using an extra pin
    pinMode(buttonHigh, OUTPUT);
    digitalWrite(buttonHigh, HIGH);

    pinMode(led1, OUTPUT);
    pinMode(led2, OUTPUT);
    setState(HIGH);
    
    IPAddress localIP = WiFi.localIP();
    sprintf(addr, "%u.%u.%u.%u", localIP[0], localIP[1], localIP[2], localIP[3]);
    Spark.variable("Address", addr, STRING);
    Spark.variable("Request", httpBuffer, STRING);
    server.begin();
}

void setState(int state) {
    digitalWrite(led1, state);
    digitalWrite(led2, state);
    ledState = state;
}

void loop() {
    delay(100);
    
    if (digitalRead(buttonInput) == LOW) {
        setState(ledState == HIGH ? LOW : HIGH);
        while (digitalRead(buttonInput) == LOW) {
            delay(100);
        }
    }
    
    client = server.available();
    if (client.connected()) {
        httpLength = 0;
        bool foundNewLine = false;
        while (client.available()) {
            char c = client.read();
            if (c == '\r' || c == '\n') {
                foundNewLine = true;
            } else if (!foundNewLine && httpLength < MAX_HTTP_LENGTH) {
                httpBuffer[httpLength++] = c;
            }
        }
        httpBuffer[httpLength] = '\0';
        client.println("HTTP/1.1 200 OK");
        client.println("Content-Type: application/json");
        client.println("Connection: close");
        client.println();
        client.println("\"ok\"");
        client.flush();
        client.stop();
        setState(ledState == HIGH ? LOW : HIGH);
    }
}
```

stevie67, 2015-08-16:
I have the same problem with my own HTTP server.

m-mcgowan, 2015-08-17:
Thanks for the test case, very much appreciated. I've made a fix for this issue in the commit above. If you're able to pull the latest develop branch and test, please confirm the issue is resolved. Thanks!

stevie67, 2015-08-17:
@m-mcgowan, I tested this with the newest develop branch and my example now works fine! Thanks for the fix!

m-mcgowan, 2015-08-17:
Thanks for confirming!

commit 3ab6ef7d (2015-08-17): close the TCP client socket using both available methods in WICED. Fixes #536.
closed 2015-08-17 by m-mcgowan