particle-iot/device-os #538: TCPClient::connect() does not close an existing socket.

andyw-lala, 2015-08-14:
TCPClient::connect() doesn't close an existing connection. It will orphan a socket if you call it without stopping the previous one. Sure, you can make the case of "Doctor, it hurts when I do this", but we should make the tcpclient (and server) classes as robust as possible, IMHO.

You will hit _this_ limit after something like 8 iterations with code similar to this:

```cpp
TCPClient client;

/*
 * Pick a host that exists, with an open http server.
 */
IPAddress server(192,168,121,51); 
int port = 80;

unsigned loop_count = 0;

void loop() {
  int x;
  
  Serial1.println("connecting to server (loop: " + String(loop_count) + ")");
  x = client.connect(server, port);
  Serial1.println("   connect returned: " + String(x));
  x = client.println("Hello World!");
  Serial1.println("   println returned: " + String(x));
  x = client.connected();
  Serial1.println("   connected returned: " + String(x));

/* With the following code, you can probably run for hundreds of connections, without it, maybe 8 or so. */
  if (x) {
    client.stop();
  }

  delay(2000);
  loop_count++;
}
```
However, it looks like there is some kind of async cleanup needs to happen in the wiced stack, because simply adding a call to TCPClient::stop() at the top of TCPClient::connect() doesn't work, whereas adding a call to client.stop(), followed by delay(2000) in application code, as shown above, does work.

If I was a betting man, I'd wager the same medicine will need to be applied to the TCPServer class as well.
~

m-mcgowan, 2015-08-17:
Totally agree, the API should make things as simple as possible and hard to shoot yourself in the foot. 

There have been many stability improvements to sockets lately (I've spent the day testing and refining), so I'm hoping simply calling stop() before connect in TCPClient/TCPServer is sufficient to avoid a resource leakage.

m-mcgowan, 2015-08-17:
For anyone watching, I believe this bug is around 2 years old from the original TCPClient implementation.

m-mcgowan, 2015-08-23:
Resolved in commit above.

closed 2015-08-23 by m-mcgowan