particle-iot/device-os #540: Death by spaces in paths on Windows

jacobq, 2015-08-15:
I'm not sure if the problem I'm having can be fixed by a change to the project or if it's just something I need to fix on my system. However, I thought it'd be worth submitting this as an issue in case others are getting hung-up by it. 

I'm trying to build the firmware from source on a Windows 7 (64-bit) system. Basically, the problem appears to be due to spaces in the paths to the tools (e.g. `C:\Program Files (x86)\GnuWin32\bin` or `C:\Program Files (x86)\GNU Tools ARM Embedded\4.9 2015q2\bin`). I have the repos downloaded in `C:\Users\Jacob\Documents\Particle\` but as you can see from the output below, running `make` fails when it hits the first `echo`.

```
$ cd ~/Documents/Particle/firmware/build/
$ ls
makefile  readme.md  tools.mk
$ make
sh: C:\Program: No such file or directory
make: *** [check_external_deps] Error 127
```

After removing the `@` prefix from lines 221-226 and running again with debug flags I see:
```
$ make --debug=bj
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i386-pc-mingw32
Reading makefiles...
Updating goal targets....
 File `all' does not exist.
   File `elf' does not exist.
      File `check_external_deps' does not exist.
     Must remake target `check_external_deps'.
echo Building core-common-lib
sh: C:\Program: No such file or directory
make: *** [check_external_deps] Error 127
```

If I try to side-step the problem by commenting out all the `echo` lines, I get another error:
```
$ make --debug=bj
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i386-pc-mingw32
Reading makefiles...
Updating goal targets....
 File `all' does not exist.
   File `elf' does not exist.
      File `check_external_deps' does not exist.
     Must remake target `check_external_deps'.
"c:\PROGRAM: Interrupt/Exception caught (code = 0xc00000fd, addr = 0x4227d3)
```

Does anyone know a way to get around this on Windows? I'm a bit daunted by it and am firing up a Linux vm and planning to build there instead. I understand Windows ' being non-POSIX sucks and is not your fault, but if anyone would be willing to share insight about how I can resolve these problems I'd appreciate your help.

technobly, 2015-08-15:
We made a change recently in the develop branch that may fix this, are you using the latest develop?
https://github.com/spark/firmware/commit/ccb900eacd0398ed2f02d6fc7999e4708a6f9d72

Also you can try converting your PATH to DOS 8.3 format.  Don't copy what you see here, you have to go through a slow process to figure out what your DOS 8.3 formatted path would look like, and then convert it for your unique system.
https://community.particle.io/t/local-toolchain-on-windows-8-1-pro-x64-error-solved/4318/6

jacobq, 2015-08-15:
Just switching branches didn't seem to make any difference. 
```
Jacob@NP-COMPLETE ~/Documents/Particle/firmware/build (origin/development)
$ git log --pretty=oneline --abbrev-commit
0001070 pointer to correct non-develop branch for photon
e28f425 pointer to Photon firmware
fe4428c Merge pull request #386 from zfields/patch-1
0030b54 Merge pull request #422 from spark/photon-warning
```

```
$ make --debug=bf
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i386-pc-mingw32
Reading makefiles...
Updating goal targets....
 File `all' does not exist.
   File `elf' does not exist.
      File `check_external_deps' does not exist.
     Must remake target `check_external_deps'.
sh: C:\Program: No such file or directory
make: *** [check_external_deps] Error 127
```

jacobq, 2015-08-15:
Oh, wait, nevermind, I see what you mean now. Yes, I changed my path by replacing "Program Files (x86)" with "Progra~2" and I can build now! Thanks for the tip!

jacobq, 2015-08-15:
See also https://community.particle.io/t/local-toolchain-on-windows-8-1-pro-x64-error-solved/4318

m-mcgowan, 2015-08-15:
great that you got it working!

closed 2015-08-15 by m-mcgowan