particle-iot/device-os #541: Print::println adds the return value from multiple print calls

monkbroc, 2015-08-17:
While debugging the socket memory leak issue, Andy had some weird test results: doing `client.println()` would normally return 14, but in error conditions sometimes -20, -48 then -3.

This is a result of `Print::println()` adding the results of multiple calls to `print` which delegates to `TCPClient::write` which then returns -1 or the result of `socket_send` which can be negative.

I'm raising the issue because if some calls to `print` return positive values and some negative values, the result might still be positive. Checking for positive values would be a sensible way to check that all characters were sent properly.

Just letting the team know. If you think that's a non-issue, you can close this.

m-mcgowan, 2015-08-17:
Thanks for raising this - I think this is an issue, since it's creating results that aren't always intuitive. 

We can either view that returning -1 from write is invalid, and fix that, or augment the Print code so that it can tolerate -1 being returned from write.  I prefer the latter since it encompasses more cases. 

E.g. writing to a socket which is subsequently closed might return a positive value for part of the message written. A subsequent write would then always return -1 since the socket is closed.

m-mcgowan, 2015-08-17:
Fix in this commit - https://github.com/spark/firmware/commit/a8912cae7f2399b5b550e8cad0a276215473547d

closed 2015-08-20 by m-mcgowan