particle-iot/device-os #542: TCPClient::connected() doesn't detect when the socket is closed

monkbroc, 2015-08-17:
[Reported by @javier-rodas inside another issue](https://github.com/spark/firmware/issues/490#issuecomment-128799318)

When a server closes the socket, `connected()` keeps returning `true`.

Test case:

```c++
TCPClient client;

IPAddress server(192,168,0,4); // Start a server with nc -l -p 8080
int port = 8080;

void setup() {
}

void loop() {
  client.connect(server, port);
  while(client.connected()) {
    if(client.available()) {
      char c = client.read();
    } else {
      delay(100);
    }
    // Closing the server doesn't break out of this loop
  }
}
```

monkbroc, 2015-08-17:
0ee169c didn't fix the test case above when only reading, or when neither reading or writing. The disconnect callback was not called.

That callback looks like it causes a lot of trouble because of the asynchronicity.

What about checking the LwIP error directly in some of the `socket_hal` functions? The key is `netconn_err`

I made a PR with this idea #543

JumpMaster, 2015-08-23:
I'm seeing the same issue with a simple web request using the develop branch.  Although the server disconnects the photon does not detect it.

    #include "application.h"
    
    TCPClient client;
    byte server[] = { 45, 56, 96, 24 }; // whatismyip.org
    unsigned long startTime;
    void setup()
    {
    	Serial.begin(9600);
        while (!Serial.available()) {
    		Serial.println("Press any key to start.");
    		delay (1000);
    	}
    
    	Serial.println("connecting...");
    	startTime = millis();
    	if (client.connect(server, 80))
    	{
    		Serial.println("connected");
    		client.println("GET / HTTP/1.1");
    		client.println("host: whatismyip.org");
    		client.println("connection: close");
    		client.println();
    	}
    	else
    	{
    		Serial.println("connection failed");
    	}
    }
    
    void loop()
    {
    	if (client.available())
    	{
    		char c = client.read();
    		Serial.print(c);
    	}
    
    	if (!client.connected())
    	{
    		Serial.println();
    		Serial.println("disconnecting.");
    		Serial.print("Time taken:");
    		Serial.println(millis()-startTime);
    		client.stop();
    		for(;;);
    	}
    }

m-mcgowan, 2015-08-28:
Tested and confirmed the test case in the issue description works. (I modified the messages to print connected when the socket is connected, and disconnected when finally disconnected.)

Thanks @monkbroc.

commit 0ee169cd (2015-08-17): fixes #542 - register a callback handler with WICED but also double check that the socket hasn't already been disposed of. There is a race condition inside WICED where the dispose callback is called after the socket has been deleted.
closed 2015-08-28 by m-mcgowan