particle-iot/device-os #544: [Core] System doesn't set color of RGB LED for SOS, etc.

WebDust21, 2015-08-17:
This applies to all builds for the Core, from Particle Dev IDE, to the RELEASE 0.4.3, DEVELOP, LATEST, etc.

Symptom: When controlling the RGB LED with `RGB.control(true);`, any system LED patterns do NOT change the color of the LED.  Say your code sets the RGB LED to green.  If your code causes a stack overflow, memory pointer error, divide-by-zero, etc., the SOS code is flashed in green, <b>not</b> red.  Ditto for holding MODE to clear WiFi credentials.  The LED blinks whatever color it has been assigned, <b>not</b> blue.

WebDust21, 2015-08-17:
NOTE: This is the same issue as #472 Develop branch only: Wifi Listen and Credentials Reset modes don't change RGB LED to blue

m-mcgowan, 2015-08-22:
Fixed!

closed 2015-08-22 by m-mcgowan