particle-iot/device-os #546: WiFi.selectAntenna reversed functionality?

mtwomey, 2015-08-17:
Building a test app with the development branch, WiFi.selectAntenna(ANT_INTERNAL) and Wifi.selectAntenna(ANT_EXTERNAL) appear to be reversed.

Test 1: WiFi.selectAntenna(ANT_EXTERNAL) set, and an antenna connected

Looping WiFi.RSSI() reports values in the -40s. Detaching or re-attaching the antenna has little impact.

Test 2: WiFi.selectAntenna(ANT_INTERNAL) set and and antenna connected

Looping WiFi.RSSI() reports values in the -20s. Detaching the antenna causes the values to jump into the -60s.

m-mcgowan, 2015-08-17:
What device are you testing this on and which commit? Thanks!

mtwomey, 2015-08-17:
Yes: commit 94bcb08fd354745846c8cb72e941e3aba68afe73

I'm using a Photon.

m-mcgowan, 2015-08-17:
There have been pre-release units made also, and we have the P1. Thanks for the info, that helps us understand the problem!

We've recently changed the antenna selection order since they appeared inverted before. @technobly is master of photon hardware and the antenna selection, let's wait for his insight.

technobly, 2015-08-17:
@mtwomey I just re-tested this and it's working as expected for me.  Please run the following to program your Photon and see if things change for you.  It sounds like you just do not have the new system firmware on your Photon.

git fetch --tags
git checkout v0.4.4-rc3
cd firmware/modules
put photon in DFU mode
make clean all PLATFORM_ID=6 -s COMPILE_LTO=n APP=yourapp program-dfu PARTICLE_DEVELOP=1

This will compile and download system-part1, system-part2 and yourapp to your photon.

mtwomey, 2015-08-18:
I just tested the above and you are correct. I was running on the develop branch before. On 4.4-rc3 (and 4.4-rc4, which I also tested) it works correctly / as indicated. Thanks!

technobly, 2015-08-18:
Good to hear, thanks!

closed 2015-08-18 by technobly