particle-iot/device-os #547: SoftAP setup should continue after user enters invalid password

m-mcgowan, 2015-08-19:
SoftAP setup not available after setup mode is automatically re-entered after failing to connect to the wifi network.

https://community.particle.io/t/strange-listening-mode-behavior/14457

m-mcgowan, 2015-08-22:
Could not reproduce this problem.

jonlogan, 2015-09-03:
Given latest comments in https://community.particle.io/t/strange-listening-mode-behavior/14457/4 and https://community.particle.io/t/photon-softap-refuses-connection-or-requires-password/15032/7 should we consider reopening this issue?  Or file a new issue?

jonlogan, 2015-09-03:
Pasting steps to reproduce as described by user @theamos in community links mentioned above, and reopening:

We're running into an issue where brand-new out-of-the-box photons refuse to allow devices to connect to their SoftAP the second time we put them into setup mode. When joining their SoftAP network it either immediately drops the client device or asks for a wifi password.

Steps to reproduce:
1. Open brand-new photon, connect to power, make sure it's in wifi setup mode
2. Using the iOS or Android Particle app, go through the SoftAP setup process to connect it to you wifi network (this works well)
3. Put the Photon BACK into wifi setup mode and go through the wifi setup process again in the app (we're using the same wifi network)
4. At the step where you connect to the Photon's SoftAP, the OS will ask for password (iOS) or the app will just hang and keep trying to connect (Android)
5. If it doesn't fail the 2nd time, it will frequently fail the 3rd time through

Any idea what's causing this issue? It's highly concerning to us since we're using stock code from Particle (no user firmware installed yet) and we essentially end up with devices that we cannot reconnect to the network no matter how many times we reset them.

jonlogan, 2015-09-04:
As a counterpoint, Mat shared these steps as a (possible?) means to circumvent this issue.  Adding for clarity.  

```I just tried again, and it's working for me. (I'm testing on a laptop connected to the Photon's SoftAP - can still contact the TCP server on port 5609.)
I'd made a mistake of not reconnecting to the SoftAP on the second try.
Perhaps the mobile is also suffering the same problem and not reconnecting WiFI to the Photon's SoftAP?
I am testing using the latest release on github - 0.4.4.rc.3. https://github.com/spark/firmware/releases/tag/v0.4.4-rc.32
Please update your system firmware to that and see if it makes a difference - this issue may have already been fixed.```

m-mcgowan, 2015-09-12:
Is resetting the device between attempts a viable workaround?

jonlogan, 2015-09-13:
I don't know, but I can ask the customer (Tank Utility / @hardwater)
tomorrow.  Thanks!

Jon Logan
Director of Customer Success
(415) 867-7878 | @jonlogan | jon@p <http://goog_773526005>article.io

HardWater, 2015-09-13:
No resetting the device is of no help. Removing power also is of no help. I would also like to add the behavior that was reason for the original  creator of this thread (no softap after wrong password) seems very similar to our problem but we are not passing the wrong password. Hope that helps.

m-mcgowan, 2015-09-13:
I'm using the latest firmware (0.4.5) and am not seeing the issue. Could you please try with the latest firmware installed. (e.g. use `particle update` to flash the photon.) Is updating the firmware a workable solution for you?

jonlogan, 2015-09-14:
Thanks Mat.  Pinging @HardWater for visibility.

jonlogan, 2015-09-30:
@m-mcgowan, some more context on this one -- another customer reported that this issue occurred for 3 of 10 devices that they have setup in the field.  

The 3 devices that exhibited this issue all attempted to connect to networks that use a captive portal (i.e. 'click here to get online', such as you see in hotels and offices).  The other 7 devices in the field connected to networks without captive portals, and did not experience this issue.  

I don't mean to imply that there is a definite correlation to captive portals here, and just wanted to share for context.

pomplesiegel, 2015-09-30:
Could someone with the means please set up a "Captive Portal" network to see if a Photon exhibits this behavior following a failed connection attempt to that network? 3/3 of our Photons running System FW 0.4.5 exhibited this issue (asking an iPhone for a password to join the "Photon-<XXXX>") network following an attempted log-on to a captive portal network.

jonlogan, 2015-10-06:
Tank Utility reports that upgrade to 0.4.5 does not appear to address this problem.  I am attempting to reproduce via iOS and a captive portal at a hotel in Minneapolis today.

jonlogan, 2015-10-06:
Adding more context from Tank Utility:

Late yesterday we have found something new and rather interesting about this problem that gives us an inkling as to what might be going on.

Once the device is in this bricked state where we cannot connect to it via softap if we walk down the street to a point where we are out of the range of the AP that it was previously connected to we can now connect to the device via softap interface.

I think this demonstrates that the device is still connected to the AP or at a minimum has partial credentials and is trying to connect to the AP while at the same time it is trying to be open to a softAP connection.

Once I walk back within range of that original AP again I can no longer connect to it via softAP.

So how to recreate this problem

with a fresh device
loop doing the following
   {
   Using an iPhone pass credentials to an AP
   once complete delete those credentials
   }

You  should be able to cause the failure in 3-4 cycles of the above
Please let us know your thoughts and what you are seeing of your end.

The wifi network in question is not a captive portal.

jonlogan, 2015-10-06:
Ok I spent the last 90 minute testing and here was my process:

- New, unboxed photon running RC4
- iPhone 6s with latest Particle app
- Started in a Sheraton hotel with captive portal
- Attempted 3 times to setup Photon via hotel captive portal network (failed each time, as expected)
- Left hotel and went to Particle office (without changing Photon's settings)
- Setup Photon on Particle office network, passed wifi credentials successfully
- Photon downloaded zero update and upgraded to 0.4.3
- Went through the following process three times:
  - Wipe wifi credentials (holding down setup button for 10 seconds)
  - Setup as new Photon, pass wifi credentials again
  - Photon restarts, LED flashes 'SOS' with a single blink (indicating hard fault)
  - However, Photon recovers, gets online, breathes cyan
- Upgraded Photon to 0.4.5 manually and attempted this again
- Went through the same process as above, but six times:
  - Wipe wifi credentials (holding down setup button for 10 seconds)
  - Setup as new Photon, pass wifi credentials again
  - Photon restarts, does NOT SOS, gets online, breathes cyan
  - (This occurred for each of the six times I tested)

Aside from the temporary hard fault in the 0.4.3 version, I'm unable to reproduce this issue either for 0.4.3 or 0.4.5, as far as I can tell.

@HardWater are there any other details about your environment that I should be aware of, or anything else you recommend I test?

HardWater, 2015-10-06:
Jon

mebrunet, 2015-10-06:
Chiming in to say that we're experiencing a similar degradation in the Photon's Soft AP usage. As we use the Phtotons (pass WiFi credentials and wipe credentials) it becomes harder to connect to their access points. Reflashing the firmware and/or resetting the devices will often allow for a connection, but the reliability is very-low (as bad as working only 1 of 5 times on some of our devices). 

I can confirm that we did not have this issue with 4.3 (as I used it extensively to develop a Soft AP over HTTP app). However we are seeing it now with 4.5.

The messages we get are "DHCP look-up failed" (from my computer), or on mobile the connection often bugs at "Obtaining IP Address..." then disconnects.

This is very concerning for us as we've already got Photon's shipped in customer products. We're happy to dedicate considerable time & energy to improving their reliability as this is a big priority for us.

jonlogan, 2015-10-06:
@mebrunet can you describe your AP setup in detail, including the brand and type of AP, bandwidths offered, etc?  

@pomplesiegel likewise, can you describe the AP setups for the devices you saw fail?  (brand and type of AP, and any additional detail you can provide; are they dual 2.4/5ghz or just 2.4; etc)

@HardWater can you isolate your Meraki and Netgear APs from each other, and test against each separately to determine how often this occurs with each type of AP?  

One interesting distinction between Meraki and Netgear is that the Meraki is a 'cloud managed AP'.  I'm not saying this is definitely the cause, but if we're able to determine that cloud-managed APs are more often / always involved when this issue occurs, it will help narrow down possible causes.

Our next step here is to find a way to reliably reproduce this so we can move forward with a fix.  We definitely acknowledge & appreciate your reports of this issue and we take them seriously.  But we've not been able to reproduce it on our side (yet), so we value whatever details and steps you can provide.  We'll work on finding a fix for this, but we definitely need to be able to reproduce it reliably first.  

cc @m-mcgowan for visibility.  Thanks everyone.

mebrunet, 2015-10-06:
Okay - we'll put together a more detailed experiment and get AP specs for you.

But to be clear, the problem on our end is with the *Photon's* AP, i.e. we can't establish a reliable connection to the Photon itself when it's in listening mode (after having connected / disconnected several times). We've had this occur with all of our laptops and phones (6 different devices).

pomplesiegel, 2015-10-14:
@jonlogan, I now have the Photons which all identically exhibited this issue at the customer field site in my hand. Back in Boston, they are now behaving **totally properly** in listening mode, allowing both my iPhone and the same iPod Touch that previously exhibited the issue to connect. 

Perhaps a Photon only exhibits this strange softAP behavior (not allowing a successful DHCP / asking an iPhone for a password) when in range of a troublesome access point it previously attempted to connect to? The WiFi creds were cleared for both of these Photons at the customer field site, but they continued to exhibit this bad behavior. 

Unfortunately I do not know the the model number / security settings of that AP, as it is in the Seattle office of Galvanize.

jonlogan, 2015-10-15:
@pomplesiegel thanks for this information.  this adds some weight to the theory that certain types of APs could be more likely to cause this issue, but others not at all.  If you are able to determine the brand and type of the AP at some point, please let us know. 

@HardWater were you able to run a parallel test between your Netgear AP and your Meraki AP?  

Thanks everyone,
Jon

jonlogan, 2015-10-15:
Also, we suggest collecting data not only on AP brand/model, but also on specific configuration; channel, security type (WPA1/2, etc), PHY mode (a/b/g/n/ac), and RSSI if possible.

With a Mac, this can be done by connecting to the same AP from the same spot as the Photon, and pressing the Alt key while pressing Network icon on Mac finder status bar.

craigfoo, 2015-11-10:
Can someone break down the process of Photon provisioning?

I want to understand how the AP provisioning works on the Photon. It seems that the universally accepted method of provisioning an IoT device is for the device to act as an AP then the user connects to it in order to send the Wifi AP credentials over via smartphone. I'm assuming this could be done via UDP or TCP.

I might be wrong but the device can't act as a AP and a Station at the same time. Correct? Going off that theory, how does the device know that the Wifi credentials are correct? It must have to shut down the AP that it creates to try and connect to the user's Wifi, right? If the credentials are NOT correct, how does it let the user know on the smartphone side? At this point, the device's AP doesn't exists and the user would have to jump back on it again. This doesn't seem to be happening when I try to provisioning my Photon through the Tinker app. What's really happening there?

TL;DR Explain the Photon Wifi provisioning process to me like I'm 5.

karlhenricksen, 2016-01-28:
Hi folks, I encountered what looks to be this issue - at some point I saw my Photon (v0.4.9) soft AP all of a sudden require a password, so I was dead in the water for most of yesterday. Here are some notes, hopefully some of it will be helpful:

1. **Main Access Point**: TP-LINK 
Firmware Version: 3.17.6 Build 20150515 Rel. 63652
Hardware Version: ArcherC5 v2
Wireless 2.4GHz
Mode:  11bgn mixed
Channel:	1,   
Channel Width:  20MHz (20/40/Auto)
WDS Status:	Disable
Security: WPA/WPA2-Personal, AES

2. **Failed Recovery**: Unable to recover with photon reset, wifi credentials cleared, reset again. 

3. **Successful Recovery**: Photon was recovered after I left in listening mode (failed state with pw requirement) overnight - when I returned the next day I was able to send credentials successfully to the Photon.

4. **Recent Environment Changes**: 
Updated to 0.4.9. Also, recently, our corporate IT group placed strong restrictions on what access points were not blocked. This one is on the white list, and we've had no other issues, but it is one change, so seems suspect.

5. **Other Notes of Importance**:
- The failure occurred after having sent 2 invalid wifi credentials via serial "wifi" command in listening mode. This amounted to the Photon going through its list, failing to connect to the first 2 on the list, and defaulting to the main access point. 
- I believe there was one other set of creds in the wifi list, which would have been the corporate network here, so it's possible it was trying to connect to that. That one is unknown and not controlled by us.
- Failure occured (I believe) after going back into wifi setup mode by holding down setup button, but not 100% on that.

@jonlogan 
Is there anything to the fact that I was entering incorrect wifi via serial that could possibly put the Photon into this state?  
Why would it clear/timeout on this error state overnight?

Looking back through this issue, it seems invalid wifi or possibly blocking due to our IT might affect this, but would appreciate any insight since I've got a situation here now that can reproduce this issue apparently.

jonlogan, 2016-02-03:
Cross-posting this issue, https://github.com/spark/firmware/issues/692, for visibility.  It may or may not apply to the behaviors described here, but it is an important reminder that hardware design could also contribute to wifi connectivity issues -- if you are testing with a device connected to your PCB, of course.

m-mcgowan, 2016-03-18:
Our current knowledge of this issue points to board design playing a key role in this bug.  Since this isn't actionable in firmware, can we close this one out?

pomplesiegel, 2016-03-18:
We're not convinced board design is the only factor to trigger this issue. 

We have seen this occur using the exact same hardware (for which we have hundreds of copies) only a handful of times. In those cases, we believe this issue (Photon in SoftAP mode requiring a password to connect) could be caused by environmental factors - possibly being surrounded by an enterprise WiFi environment configured in a certain manner. 

This is something we'll certainly be watching for as it has already occurred once on a customer site.

mebrunet, 2016-03-20:
I'm comfortable associating it with board design and closing the issue.
However, in our team's humble opinion it remains something to be improved
on Particle's hardware end. We've nonetheless seen this issue arise when
following the Photon's keep out zone design requirements, and power specs -
only an external antenna and a quality 1000mA power supply seem to 100%
effective.

technobly, 2017-02-07:
There were a number of SoftAP bugs fixed in later firmware revisions and this problem as described in the community post appears to be resolved for 0.6.1-rc.2.  I have not tested all versions in-between that and really old 0.4.x though.  Feel free to reopen if the problem persists in newer firmware.

closed 2017-02-07 by technobly

peternoyes, 2017-06-12:
We are experiencing this issue with both 0.6.1 and 0.6.2

In our workplace we have Meraki access points. 

- We use the SoftAP functionality and iOS to configure a P1 and it works the first time. We then Reset the device in order to re-test the claiming process (By holding down the reset button for 10+ seconds). iOS will prompt for password any time we try to connect to the P1 via WiFi. 
- If we wait a day or so then it suddenly will work again, but just once.
- This is 100% reproducible. Happens on every single device we try.

HardWater, 2017-06-13:
Hi Peter,

Here is my understanding of what is going on Meraki routers are on the lookout for devices termed rogue APs. A device that connects to the Meraki router and offers other devices access to it can effectively break the security setup at the router. So what is happening is the Meraki router is clonin the details of the AP being broadcast by the Particle device and offering it up as a competing AP with same ssid. The reason you are seeing the password request is that you are actually connectted to is the cloned AP not the Particle device itself.

peternoyes, 2017-06-13:
I now see what is happening. The feature in Meraki units that does this is called "Air Marshal." It is triggered by the device both having been an access point _and_ subsequently being connected to the network, which is why it works the first time. It won't shadow the access point SSID if it never goes on the network.

If the Photon could somehow use different Mac addresses in setup mode vs when it connects to the network then this problem wouldn't happen.