particle-iot/device-os #558: 0.4.4 Listening mode re-enters listening mode after credentials are given

LeoWW, 2015-08-21:
Steps to reproduce issue:
- git pull from "develop" branch of spark/firmware
- run any user app on photon 
- go into listening mode (I did this by calling wifi.listen() in user code, but it should probably work by pressing setup for 3 seconds as well) -- photon is now blinking blue
- enter credentials using the particle app
- observe issue -- photon will blink white and then proceed to blinking blue instead of blinking green (which is the expected behavior) which indicates that the photon accepted the credentials but didn't try to connect to wifi and went back into listening mode instead

m-mcgowan, 2015-08-21:
Does hitting reset help?

LeoWW, 2015-08-21:
If you hit reset after it has credentials it will connect.
If not, you enter reset mode again.
We want the photon to connect immediately after receiving credentials due to our use case. 

Use Case:
-- Customer uses our app while the photon is in listening mode, enters credentials, and expects the device to connect to wifi. Customer has no knowledge of the photon device so the user won't be able to hit reset.

m-mcgowan, 2015-08-21:
I've found the fix, which requires a small change to system firmware, which will happen in the next scheduled release. 

For now, there is a workaround you can add to your application:

```
void system_event_handler(system_event_t event, uint32_t value, void* data)
{
    if (event & wifi_listen_end) {
        WiFi.listen(false);
    }
}

void setup()
{
    System.on(all_events, system_event_handler);
}
```

This will prevent re-entry to listen mode.

m-mcgowan, 2015-08-21:
Confirmed fixed.

closed 2015-08-21 by m-mcgowan