particle-iot/device-os #560: PHOTON: WiFi.SSID() may not return the network currently connected to

pomplesiegel, 2015-08-21:
When `WiFi.SSID()` is set as a gettable char* OTA 

```Spark.variable("wifiSSID", WiFi.SSID(), STRING);```

and a Photon successfully switches networks due to a call to `WiFi.setCredentials()`, it looks like WiFi.SSID() is not updated to reflect the new network it's now connected to.

In our case, we called `WiFi.setCredentials()` to place the Photon on a new network, then unplugged the old router. The Photon was happily connected to the new network, so all OTA `get`s worked fine, but the value of `wifiSSID` reflected the old network.

pomplesiegel, 2015-08-21:
Could be part of a larger issue: When the router was again removed and the Photon jumped to another saved network in range, it shows stale information for both `WiFi.SSID()` and `WiFI.IocalIP`. However, it is definitely successfully connected to the new network (breathing cyan and responsive OTA)

m-mcgowan, 2015-08-21:
I'm a bit confiused - `WiFi.setCredentials()` on it's own won't move the photon over to another AP. Can you be sure it was connected to the new AP?

pomplesiegel, 2015-08-21:
Thanks for the response. The Photon was definitely on the new AP, as the old one was unplugged from power completely, and there are no other APs in range with its name. Our code automatically calls

```
  WiFi.off();
  delay(500);
  WiFi.on();
  WiFi.connect();
  Spark.connect();
```

after it has lost an internet connection and failed a ping. In this case, that caused it to seek out an alternative AP, which it did find and connect to. However, after forming this connection it shows old information for both the SSID and IP.

m-mcgowan, 2015-08-21:
Thanks for that extra detail. I'll attempt to reproduce the problem here and report back with any fixes/workarounds.

m-mcgowan, 2015-08-21:
It will take me some time to setup a test for this (which I'll do in due course). If you're able to build locally, please pull develop and try the fix above! Thanks!

closed 2015-08-26 by m-mcgowan