particle-iot/device-os #561: PHOTON: `WiFi.hasCredentials()`not TRUE following `particle setup wifi`

pomplesiegel, 2015-08-21:
Within our program, which runs in `SYSTEM_MODE(MANUAL)`, `setup()` has a quick check 

```
//if no creds found, enter listening mode
if( !WiFi.hasCredentials() ) 
  WiFi.listen();
```

With this line in place, Photon re-enters listening mode directly following a successful call to `particle setup wifi`. 

The system LED goes green for a quick moment following a reset (as if the Photon will now attempt to connect to a network), but quickly re-enters listening mode. 

Removing the check allows successful connection to a stored network.

pomplesiegel, 2015-08-21:
Likely related to #558

m-mcgowan, 2015-08-21:
@pomplesiegel  can you try the workaround in 588 in your application, or alternatively build system firmware from develop, which has the fix?

pomplesiegel, 2015-08-21:
Sure!

pomplesiegel, 2015-08-21:
Yes, the fix from #558 fixed this issue. Closing now. Thanks!

closed 2015-08-21 by pomplesiegel