particle-iot/device-os #562: PHOTON: Connections to hidden SSID networks?

pomplesiegel, 2015-08-21:
On the Core we were able to connect to networks with hidden SSIDs by explicitly calling `WiFi.setCredentials()` with the exact SSID and password of the network. Photon seems to be flashing green when attempting to connect to these hidden networks. When the network is changed to broadcast its SSID and the Photon is restarted, it happily joins the network. 

Running in `SYSTEM_MODE(MANUAL)`. Network in question is secured w/ WPA2 AES.

m-mcgowan, 2015-08-22:
The main issue here is that WICED on the photon requires the cipher type in order to connect. With a broadcasting network, the device is able to scan the local APs and detect the cipher type. It's not clear to me how we could achieve the same thing when connecting to a non-broadcasing AP.

pomplesiegel, 2015-08-22:
I see! I am able to migrate a Photon onto a hidden network using the iOS app and soft-AP functionality. Perhaps this is not necessary, just more of a nice to have, given Core inherently has this functionality.

m-mcgowan, 2015-08-22:
That's interesting. Do you have to select the BSSID (MAC address) of the hidden network?

pomplesiegel, 2015-08-22:
Using the iOS app, we just follow the normal steps, but choose to enter our network name manually. After entering the network name manually, the next page prompts you to enter your password. At this stage the app displays the cipher type (in our case, AES) and only asks for the password. Perhaps this is gathered from a previous connection on the iPhone. Not sure.

m-mcgowan, 2015-08-30:
I pushed a fix in the PR referenced above, if you're able to test. (Check out the `feature/hidden-ssid` branch, and build and flash `module/photon/system-part2`)

With this change, serial setup of the Photon works with hidden and offline networks.

pomplesiegel, 2015-09-02:
Hi @m-mcgowan,

I compiled locally using `make clean all PLATFORM=photon` and produced the binary in `build/target/system-part2/platform-6-m/system-part2.bin`.

When flashing this either OTA or using ````dfu-util -d 2b04:d006 -a 0 -s 0x8060000:leave -D system-part2.bin```` my photon is blinking red. When i tried flashing it using `particle flash --usb system-part2.bin` it booted up ok,  but i'm not sure if that flash actually worked.

reverting to 0.4.4 via `particle update` fixes the red blinking. 

Anything I'm doing wrong?

m-mcgowan, 2015-09-02:
Could you also flash system-part1 first please, there may have been a change there that I'd overlooked.

pomplesiegel, 2015-09-02:
Just tried flashing system-part-1 and still red flashing.

It appears it's actually linked to a combination of our user FW and the new system FW. However, our FW works ok w/ 0.4.4.

When loading `tinker` everything's OK (breathing cyan), but it's difficult to say which network the Photon is on without our own code running.

Tried building our code locally w/ this branch to see if that was the red flashing issue, but i'm getting ````undefined reference to `Wire'```` all over the place.

What i'm hoping to verify 
1. Do particle serial commands work for entering wifi creds, when a network is unavailable or hidden?
2. Does setCredentials() now have the same behavior, without you having to know the cipher? Should be possible, as if the network is offline, particle-cli wouldn't know the cipher either, correct?

m-mcgowan, 2015-09-02:
If the network is offline/hidden then the cipher needs to be provided by the user, either via setCredentials, or via serial/particle-cli. 

CLI support is being added as we speak! 

To remove the problem with undefined references, please try a clean build. 

I'm a little worried about your firmware SOS'ing. If you're able to share the source, I'd love to investigate.

wmjenk, 2015-09-05:
@m-mcgowan Unfortunately, using the iOS Particle app does not work for a hidden SSID with no password. The post above suggested it did but the person also previously connected to the same network when it wasn't hidden. See the Oops! screen below that pops up when you try to connect to a hidden SSID with no password. I see that a fix is in the works (any idea when it might be available in released firmware?) when using the CLI but it would be nice if it was also fixed in the Particle iOS app since this is a more common method for setting up Wifi (especially if they are customers for a product based on the Photon).

[image]

m-mcgowan, 2015-09-12:
This is resolved in 0.4.5!

closed 2015-09-12 by m-mcgowan

pomplesiegel, 2015-09-14:
Hi @m-mcgowan, now realized we were SOS'ing because we included the redundant workaround `void system_event_handler()`. All seems well in 0.4.5!