particle-iot/device-os #564: Add a "Recipes" section to build.md with common commands

monkbroc, 2015-08-21:
There are many variations of make commands based on the current folder and build flags. It's hard to know why combinations work and are useful.

Having a section called "Recipes" in [build.md](https://github.com/spark/firmware/blob/develop/docs/build.md) listing some common commands would be great. Put commands you use often in the comments and then I can make a PR.

Here are some of my frequent commands:

```
# Complete rebuild and DFU flash of latest firmware
firmware/modules$ make clean all program-dfu PARTICLE_DEVELOP=1 PLATFORM=photon

# Incremental build and flash of firmware
firmware/modules$ make all program-dfu PARTICLE_DEVELOP=1 PLATFORM=photon

# Incremental build and flash user application.cpp only
firmware/main$ make all program-dfu PARTICLE_DEVELOP=1 PLATFORM=photon

# Build for use with debugger (Programmer Shield)
firmware/modules$ make clean all program-dfu PARTICLE_DEVELOP=1 PLATFORM=photon USE_SWD_JTAG=y

# Build with external application with stable firmware (latest branch)
firmware/modules$ make all PLATFORM=photon APPDIR=~/my_app

# Rebuild and flash on of the test application
firmware/user$ make clean all program-dfu TEST=wiring/no_fixture PLATFORM=photon PARTICLE_DEVELOP=1

# Build the compilation test (don't flash on device)
firmware/user$ make TEST=wiring/api PLATFORM=photon PARTICLE_DEVELOP=1
```

m-mcgowan, 2015-08-21:
nice idea. I put PARTICLE_DEVELOP in my environment so it's not needed on every make invocation.

monkbroc, 2015-08-21:
Good thing to put in a Recipes section. I didn't realize that would be equivalent :-)

m-mcgowan, 2015-08-21:
I've wanted to do this, but equally I think everything covered here is in the build docs. 

My main hesitation is that we will soon be supporting 3 platforms and surely more to follow - do we want to have recipes in triplicate - for each platform?

m-mcgowan, 2015-08-21:
I go back on that last comment, I think having a Photon recipes section would be good. Just to summarize the info already provided in the various sections of the build docs. Especially since it's more complex than the core with separate system/application.

monkbroc, 2015-08-21:
A full factorial of examples is not necessary. I think that having various example that accomplish specific goals are sufficient. I can extrapolate that I can change the platform for example, but it was hard for me to figure out the combination to recompile and flash only the user application.

technobly, 2015-08-21:
This is a great idea.  Just thinking about this PARTICLE_DEVELOP thing some more, it's going to be kind of silly to put that in examples if the purpose of it is to "block" users from accidentally using the develop branch.  If we want to keep it a bit tricky, we should omit that... and the message that pops up when it's not included should clearly indicate what to type to continue compiling if users REALLY want to.  Not all users are going to know what "export" means.

m-mcgowan, 2015-08-21:
Building from sources, particularly from develop should be considered an Advanced technique, and knowledge of `export` a necessary prerequisite.

monkbroc, 2015-08-21:
>  the message that pops up when it's not included should clearly indicate what to type to continue compiling if users REALLY want to

I disagree with that. The message doesn't actually say where the build documentation is.

I hit a dead end when I first tried to compile locally. My steps were: `git clone`, `git checkout develop`, `make`. I was then shown this:

>*** Please note the develop branch contains untested, unreleased code. We recommend using the 'latest' branch which contains the latest released firmware code. To build the develop branch, please see the build documentation..

This would be better:

>*** Please note the develop branch contains untested, unreleased code. We recommend using the 'latest' branch which contains the latest released firmware code. To build the develop branch, please see https://github.com/spark/firmware/blob/develop/docs/build.md

m-mcgowan, 2015-08-21:
Agreed. I added a direct link to the message. On the one hand, I don't want this to be too easy, but really the main point was for users to read that message and decide for themselves if they really mean to build this branch or switch to latest.

m-mcgowan, 2015-08-21:
I'll close this since the main point has been addressed. If any tweaks or updates are needed, we can continue those in a PR. Thanks.

closed 2015-08-21 by m-mcgowan