particle-iot/device-os #565: [Photon] Stuck in fast green blinking

pomplesiegel, 2015-08-21:
One of our photons (running 0.4.4 in `SYSTEM_MODE(MANUAL)` ) just disconnected from WiFi and began blinking a very fast green, while `loop()` was totally blocked. This went on for minutes, only resolved by a reset.

m-mcgowan, 2015-08-21:
Fast blinking green means that WICED is obtaining and IP address from the router via DHCP. There is code to timeout after 15 seconds, although it's not clear why this didn't work for you.

pomplesiegel, 2015-08-21:
Yes, it's usually a temporary blink, but in this case it was indefinite. We'll keep watching for this.

m-mcgowan, 2015-09-21:
Any more info to share about this issue? If you have steps to reproduce that would help us troubleshoot.

pomplesiegel, 2015-09-22:
Hi @m-mcgowan, Haven't seen this again using 0.4.4 or 0.4.5. Probably safe to close?

m-mcgowan, 2015-09-22:
Thanks. Let's close for now and re-open if it happens again.

closed 2015-09-22 by m-mcgowan

lemontongs, 2015-10-01:
I'm seeing this issue. Or something similar. I can't tell if its blinking green or white but it is blinking quickly and indefinitely after about a day of working correctly. I have attached my code for reference. I am building for 0.4.5. Only difference from OP is that I am not using manual mode.

```

const int NUM_ZONES = 8;
const int UDP_PORT  = 5100;
IPAddress multicast( 225, 1, 1, 1 );

UDP server;

uint32_t new_zone_state = 0;
uint32_t old_zone_state = 0;
uint32_t shift = 0;

String msg;

void setup()
{
    // Set the pin modes to INPUT_PULLDOWN
    for ( pin_t pin = 0; pin < NUM_ZONES; pin++ )
    {
        pinMode( pin, INPUT_PULLDOWN );
    }
    
    server.begin( UDP_PORT );
}

void loop()
{
    // Read the pin states into the state array
    new_zone_state = 0;
    shift = NUM_ZONES - 1;
    for ( pin_t pin = 0; pin < NUM_ZONES; pin++ )
    {
        new_zone_state += ( digitalRead( pin ) << shift );
        shift--;
    }
    
    //
    // Check for a change in state, or send if 30 seconds has elapsed
    //
    bool send_signal = ( new_zone_state != old_zone_state );
    
    if ( millis() % 30000 == 0 )
    {
        send_signal = true;
        
        // Restart the UDP server for good measure
        server.stop();
        server.begin( UDP_PORT );
    }
    
    //
    // Send the new state
    //
    if ( send_signal )
    {
        msg = "0000000000" + String( new_zone_state, BIN );
        msg = msg.substring( msg.length() - NUM_ZONES, msg.length() );
        
        server.beginPacket( multicast, UDP_PORT );
        server.write( msg );
        server.endPacket();
    }
    
    // Save the new state for comparison
    old_zone_state = new_zone_state;
}
```

m-mcgowan, 2015-10-01:
Can you reduce this code to the minimum that still produces the problem? Does the problem disappear after reset? Is the WiFi router still available with the same credentials?

karlhenricksen, 2016-07-28:
Hi @m-mcgowan @technobly I believe I just saw this same issue with 0.5.3rc1 using tinker. Have you seen anything like this with photons left connected to wifi? I suspect it has to do with sometimes-inconsistent wifi here but haven't done much testing to confirm...

**Here is the chain of events leading up to the issue** (luckily I recorded, since I was testing for another issue):
**Setup:** Tinker, Sys v0.5.3rc1, 3 different photons in setup
* Left all 3 photons without wifi credentials running for 4-5 days (blinking blue)
* Then for each Photon, I followed these steps:

1. Send incorrect wifi credentials - returns to blinking blue
2. Resend correct wifi credentials _for the same router_ - connects and goes to breathing cyan

* For one Photon, I saw it fall offline temporarily, but then never returned to online state

**Results:**
It remained blinking green for maybe an hour before I realized it was not going to get back online. 
- I tried putting into listening mode and noticed it was unresponsive [(click for video)] (https://youtu.be/OCMw0Tdw4Ek)
- Reset was the only way to recover

JFdeB, 2017-09-20:
Photon rapidly blinking green.
I have a similar problem, the photon is blinking rapidly green and unresponsive, program is stuck.
The photon is supposed to do a small task and go to sleep for 10 minutes. Battery will last for months, but when the photon gets stuck the battery drains in a day. This happens after the program has run for weeks.
The problem started when i flashed the 0.6.1 firmware.
My code is below, the main problem is that there is no time out on connecting the WIFI. If that were the case, the execution of the loop would fail, photon would go to sleep and try again the next time.
CAN you check the WIFI connect code and make sure the timeout is always working?

SYSTEM_THREAD(ENABLED);
STARTUP(WiFi.selectAntenna(ANT_INTERNAL));

#include "Adafruit_BME280/Adafruit_BME280.h"
#define SEALEVELPRESSURE_HPA (1013.25)
#include "PowerShield/PowerShield.h"

PowerShield batteryMonitor;

void setup() {
    
}

void loop() {
    //    float temperature;
    String messageString;
    Adafruit_BME280 bme;
    if ( !Wire.isEnabled() ) {
        Wire.begin();
    }
    //    Wire.begin();
    delay(1000);
    
    batteryMonitor.reset();
    batteryMonitor.quickStart();
    
    bme.begin();
    
    delay(1000);

    float cellVoltage = batteryMonitor.getVCell();
    float stateOfCharge = batteryMonitor.getSoC();
    float bme_t = bme.readTemperature();
    float bme_h = bme.readHumidity();
    float bme_p = bme.readPressure() / 100.0F;
    
//    WiFi.on();
    waitFor(Particle.connected, 30000);
    if (Particle.connected()) {
        //    messageString = String(cellVoltage) +',' + ' ' + String(stateOfCharge);
        //    Particle.publish("ps-soc", messageString, 60, PRIVATE);
        bme_t = bme.readTemperature();
        bme_h = bme.readHumidity();
        bme_p = bme.readPressure() / 100.0F;
        messageString = String(cellVoltage) +',' + ' ' + String(stateOfCharge) +',' + ' ' + String(bme_t) + ',' + ' ' + String(bme_h) + ',' + ' ' + String(bme_p) + ',' + ' ' + String(WiFi.RSSI()) + ',' + ' ' + String(WiFi.SSID());
        // messagestr = cellVoltage, stateOfCharge, mcp_t, mcp2_t, bme_t, bme_h, bme_p, dht_t, dht_h, WiFi_RSSI, WiFi_SSID
        Particle.publish("temp-hum", messageString, 60, PRIVATE);
        //Particle.publish("acknowledge", messageString, PRIVATE, WITH_ACK);
    }
    
    delay(5000); //extra delay added after messages were not revceived March 23, 2017. Problem of v0.6.1
    
    //  loop takes 7 seconds
    //    System.sleep(SLEEP_MODE_DEEP, 53);
    //    delay(53000);
    System.sleep(SLEEP_MODE_DEEP, 593);
    
}