particle-iot/device-os #566: [Photon] holds constant green when attempting to enter Listening Mode

pomplesiegel, 2015-08-21:
When Photon is running an application in `SYSTEM_MODE(MANUAL)`, and not yet connected to a network (which may not be in range),if the user holds down the mode button to enter listening mode, the Photon turns solid green for a few minutes. It eventually enters listening mode after the delay.

m-mcgowan, 2015-08-21:
Do you know what your app is doing at the time? Something is blocking the app code from entering listening mode. LED freezes after 3 seconds, because the listening flag is set, which stops the normal LED handling routine from updating the LED, since it's handled by the listening routine, but the listen routine hasn't started because application code is presently running something else. 

This will be fixed with multithreading support.

pomplesiegel, 2015-08-21:
App is most likely interfacing w/ other i2c peripherals at the time. Yes, I'd imagine many of these issues will be fixed by multithreading :)

anthonywebb, 2015-08-21:
@m-mcgowan I have this same issue, simple script to duplicate:
https://gist.github.com/anthonywebb/fe2b5958c6d42f07d6aa

I am trying to allow my device to enter listening mode if the creds it has stored are not able to connect, but only stay listening for a wile, then restart everything again.

This worked in 4.x, so it is something new to 4.4

m-mcgowan, 2015-08-22:
@anthonywebb I'm not sure what you're reporting is the same problem. Here it's talking about a slow/delayed entry into listening mode, where as you're concerned with listening mode not sticking. Please reconsider this and if you see they are different please create a new issue. If they are not different, please explain further. Thanks.

LeoWW, 2015-08-31:
Can confirm that this issue occurs when flashed with app code. 
Interesting observation, though, is that when I try this in safe mode with app code the photon LED also turns solid green.
However, if you do a factory reset, you can enter listening mode without turning solid green (once credentials have been given, otherwise the photon just starts up in listening mode).

LeoWW, 2015-09-10:
Any updates on the fix/multi-threading support @m-mcgowan? The solid green state used to take around 30 seconds but now may last longer than a minute before it actually enters listening mode. Same thing happens when in safe mode with user code... is there a quick way to get into listening mode?

m-mcgowan, 2015-09-13:
It's something blocking in the WICED library. I've added already two ways to unblock but there may be other loops/delays that block the main thread. If you can provide details on exactly when it blocks vs doesn't block that would help. For example, how many WiFi networks are configured, how many are available, how repeatable is the blocking behavior - sometimes, often, each time.  This information will help me diagnose and fix the problem.

m-mcgowan, 2015-09-21:
Anyone that has access to the develop branch, please pull the latest code and test. The issue of taking a long time to enter listening mode should now be resolved.

closed 2015-09-27 by m-mcgowan