particle-iot/device-os #574: [Photon] `WiFi.setCredentials` only works for networks currently in range?

pomplesiegel, 2015-08-22:
While Photon was connected to our office network, I called `WiFI.setCredentials()` to hard-code a customer's WiFi creds into it before it went to the customer's home. 

Following `setCredentials` our code running in `SYSTEM_MODE(MANUAL)` calls 

````cpp
  WiFi.off();
  delay(500);
  WiFi.on();
  WiFi.connect();
  Spark.connect();
````

At this point it disconnects (flashing green), then connects back to the office network (still stored), as the new customer creds do not represent a network currently in range.

However, when the Photon was put in range of the customer's AP, it still did not connect (instead flashing green). When the customer mimicked our office network using an iPhone hotspot and I again called  `WiFI.setCredentials()`with the network in range, it successfully connected to the customer's AP. 

Does the Photon ignore/erase creds if they do not work upon the first try?

m-mcgowan, 2015-08-22:
Yes, just as for the other issue with hidden networks -  `setCredentials()` must be able to scan the network to obtain all of the network details.

It would be possible to extend `setCredentials()` so that these details are passed in as part of the function call. Would that help?

pomplesiegel, 2015-08-22:
Sure, that would help! This is mainly crucial on a dev level, before we have the soft-ap migration API integrated into an app.

m-mcgowan, 2015-08-22:
If you pull `develop` you'll be able to call
```
 WiFi.setCredentials("ssid)",4,"password", 8, WPA2, WLAN_CIPHER_AES);
```
Which doesn't scan the network to find the cipher, so the AP can be added offline.

pomplesiegel, 2015-08-24:
@m-mcgowan, this passed w/ flying colors on a real test. Thanks!

m-mcgowan, 2015-08-24:
Cool. Thanks for confirming!

closed 2015-08-24 by m-mcgowan