particle-iot/device-os #580: strange interaction between SYSTEM_MODE and System.sleep

bprobbins, 2015-08-24:
There seems to be a strange interaction between SYSTEM_MODE and sleep that goes beyond the sleep bug, as evidenced by the variations I tried in the code below:

Testing  System.sleep(D3, RISING, 5)
***develop branch (8/24/15 10am)
--SYSTEM_MODE(AUTOMATIC) 
  photon appears inactive (doesn't blink) after 60 second powercycle (unplug for 60 secs, then reapply power)

--SYSTEM_MODE(SEMI_AUTOMATIC)
  photon appears inactive after 60 second powercycle

--SYSTEM_MODE(SEMI_AUTOMATIC) DECLARED ONLY IN SETUP BLOCK
(I know this is "illegal" but it works!)
  photon DOES start and blink after 60 second powercycle

***latest branch (8/24/15 10am)
--SYSTEM_MODE(AUTOMATIC) 
  photon appears inactive after 60 second powercycle

--SYSTEM_MODE(SEMI_AUTOMATIC)
  photon appears inactive after 60 second powercycle

--SYSTEM_MODE(SEMI_AUTOMATIC) DECLARED ONLY IN SETUP BLOCK
  photon DOES start and blink after 60 second powercycle

#include "application.h"
//SYSTEM_MODE(SEMI_AUTOMATIC);
uint32_t lastTime;

void setup()
{
  pinMode(D7,OUTPUT);
  unsigned start = millis();
  while ( (millis() - start) < 1000) {Spark.process();}//while ( (millis() - start) < 10000)
//  SYSTEM_MODE(SEMI_AUTOMATIC);
}//setup

void loop()
{
    Spark.process();
    System.sleep(D3, RISING, 5);
    digitalWrite(D7,HIGH);
    delay(200);
    digitalWrite(D7,LOW);
}//loop

m-mcgowan, 2015-08-24:
Thanks for the report. Can you confirm you're using the latest system firmware?

The issue linked above (480) isn't related to this since it doesn't enter stop mode, code continues executing and turns off wifi for the specified duration, whereas System.sleep(D3, RISING, 5) stops execution until D3 rising or a 5 second timeout.

bprobbins, 2015-08-24:
I freshly git cloned firmware, then git checkout latest (with a git pull for good measure),  and compiled from modules. Same for testing from develop

m-mcgowan, 2015-08-24:
:+1:  Thanks, I'll take a quick look at this later today.

m-mcgowan, 2015-08-25:
Reproduced the sleep issue with a simple test app:

```
#include "application.h"
SYSTEM_MODE(MANUAL);
void setup()
{
    System.sleep(D3, RISING, 2);
    RGB.control(true);
    RGB.color(255,255,0);
}
```

Behavior:
- breathes white led for a few seconds
- led shuts off
- device does not resume after 2 seconds as expected
- touching any other pin wakes the device and the LED is steady yellow.

Expected behavior:
- breathes white led for a few seconds
- led shuts off for 2 seconds
- LED is steady yellow

@satishgn  Could you please look into this. Thanks!

m-mcgowan, 2015-09-14:
@bprobbins  this appears to be working now. Would you mind re-testing and confirming the fix with 0.4.5 firmware? Many thanks!

bprobbins, 2015-09-14:
@m-mcgowan - Yes, I can confirm that under release/v0.4.5 it now behaves appropriately on cold power up. Thanks!
Also, issue https://community.particle.io/t/core-interesting-results-with-manual-mode/13435/3 appears fixed as well. Double thanks!

m-mcgowan, 2015-09-14:
Great to hear - thanks for verifying :+1:  closing out!

closed 2015-09-14 by m-mcgowan