particle-iot/device-os #581: SyncTime request should account for device->cloud latency

dmiddlecamp, 2015-08-24:
If we started a clock before we make the request, and measured the elapsed time when we receive the response, then we should set the clock to:

```
# pseudocode follows:
unsigned int elapsedMillis;  // time it took for the whole roundtrip exchange
unsigned int latency = elapsedMillis / 2;
time = serverTime - latency;
```

Thanks!
David

m-mcgowan, 2015-08-24:
I see this would make the time more accurate than it is now, but still not totally correct since it doesn't account for any processing delay on the server or cases where the upstream latency is different from the downstream.

Should we simply use NTP for this?

dmiddlecamp, 2015-08-24:
This should get us in the realm of sub-second accuracy, and there is also an NTP library people could use.  I think this basic adjustment would get us a long way.

m-mcgowan, 2015-08-26:
Test app:

```
void setup()
{
    Serial.begin(9600);
}

/* This function loops forever --------------------------------------------*/
void loop()
{
    if (Serial.available())
    {
        char c = Serial.read();
        if (c=='t') {
            Serial.println(Time.timeStr());
        }
        if (c=='s') {
            Particle.syncTime();
        }
    }
}
```

Test procedure:
- flash the app to the device
- open http://time.is
- press 't' in tandem with the time changing
- verify the time is the same or behind by no more than 1 second. 

Since time is stored to a precision of 1 second, the time reported can be up to one second behind. Improving this requires the protocol be extended to send a timestamp with greater precision.

closed 2015-08-26 by m-mcgowan