particle-iot/device-os #589: Make attachInterrupt() not require pinMode(INPUT)

zsup, 2015-08-28:
See this pull request on the docs:

https://github.com/spark/docs/pull/148

The change to the docs is a good short-term fix, but long-term it'd be better to be more compatible with Arduino and eliminate the requirement for setting the pin as an input before attaching an interrupt.

cc @geoffsharris

m-mcgowan, 2015-08-29:
Nice catch.  I wonder if the fix is as simple as checking the `pinMode` inside `attachInterrupt`, and if it's not one of the input modes (`INPUT`, `INPUT_PULLDOWN`, `INPUT_PULLUP`) then set it to `INPUT`?

A little wary of that right now since we had this recent issue - https://github.com/spark/firmware/issues/585, which has apparently resolved itself, although the workaround was to set the pin mode to `OUTPUT`. I'll write some unit tests for both cases to see what's happening in practice.

ScruffR, 2015-10-08:
Please add an optional parameter to allow for alternative pinModes (including OUTPUT ;-) ), since there might be external pull resistors inplace anyhow, or you've got sensors that don't need pull-resistors but I still don't want a CHANGE trigger.