particle-iot/device-os #594: In manual mode WiFi.connect and Particle.connect try only once.

m-mcgowan, 2015-09-03:
If a  connection fails. it should not be reattempted.

ScruffR, 2015-10-30:
Related issue #357

m-mcgowan, 2016-03-18:
I'm concerned about changing behavior, so would like to keep the default behavior as is.  The `Particle.connect()` API can be extended to include number of connection attempts: e.g.

```
Particle.connect(5);
// will attempt to connect to the cloud 5 times.
```
This can be supported in any system mode - even automatic mode.

Internally, what happens is that once the number of connection attempts has been tried, the system no longer tries to connect to the cloud. The uin8_t `SPARK_AUTO_CONNECT` flag could be used as a counter - the cloud will try connecting to the cloud so long as this is non-zero, decrementing the counter with each connect attempt.  The value 255 would mean always try to connect (the current behavior) and the value is not decremented.

ScruffR, 2016-03-18:
An extra parameter to supply an amount of seconds when to retry after x attempts have failed might be good too.
i.e. Try five times, wait 600 sec and try five times again.

sergeuz, 2016-04-01:
What if we just add optional flag argument to `connect()`, e.g. `Particle.connect(NO_RETRY)` and later allow retrying policy to be configured globally: max. number of attempts, timeouts, linear / exponential intervals, etc?

m-mcgowan, 2016-04-08:
That could work too.  I'm drawn more towards putting everything in the Particle.connect() call rather than having both parameters to `Particle.connect()` and global flags, since then you have to look in two places to understand what's happening.  

A timeout might be more intuitive than number of connection attempts. It wouldn't be absolute, but merely a hint - "if the elapsed time since Particle.connect() was called is greater than the timeout then don't connect".  It gives users some idea how long their application code is blocked for when the system connects to the cloud. 

I also like the NO_RETRY flag (1 connection attempt) since it's allowing the application developer to say - "try to connect once and give me control as soon as possible".

ScruffR, 2018-07-19:
Has this dropped off the table?
https://community.particle.io/t/particle-connect-disconnects-wifi-can-this-be-stopped/43089/4