particle-iot/device-os #595: Pin A0 cannot be used as output, because it is configured as wake up pin by default on the core

elcojacobs, 2015-09-04:
Since this commit in March https://github.com/spark/firmware/commit/0209f247ade2ead2bf496550fe025fda126c9307 pin A0 cannot be used as digital output.

I have 4 LEDs on the BrewPi board, for the 4 SSR outputs. This simple test APP cycles through these outputs.

``` c++
/* Includes ------------------------------------------------------------------*/
#include "application.h"

SYSTEM_MODE(SEMI_AUTOMATIC);

// Define the pins we're going to call pinMode on
int act1 = A0;
int act2 = A1;
int act3 = A6;
int act4 = A7;
int buzz = A2;
const bool invertBuzzer = false;

void setup() {
    pinMode(act1, OUTPUT);
    pinMode(act2, OUTPUT);
    pinMode(act3, OUTPUT);
    pinMode(act4, OUTPUT);
    pinMode(buzz, OUTPUT);
    digitalWrite(buzz, !invertBuzzer);
    delay(200);
    digitalWrite(buzz, invertBuzzer);
}

void loop(){
    int actuators[4] = {act1, act2, act3, act4};
    for(int i=0;i<4;i++){
        digitalWrite(actuators[i], HIGH);
        delay(500);
        digitalWrite(actuators[i], LOW);
        delay(500);
    }
}
```

If I comment out the line (revert the commit)
```
/* Enable WKUP pin */
PWR_WakeUpPinCmd(ENABLE);
```
Then pin A0 toggles fine. With the line present, the pin does not toggle.

I think this should be resolved by either:

- Not configuring the pin as wake up pin by default
- Changing the pin configuration when pinMode is called on A0.

This problem is not present on the Photon. On the Photon, the wakeup pin is A7, but this pin can be used as digital output without problems.

m-mcgowan, 2015-09-10:
I tested prior to this fix and couldn't make wakeup work on a core with the latest bootloader. And WKP doesn't work after this fix either, but A0 works as expected. So keeping the fix and we will investigate WKP on the core later.

closed 2015-09-10 by m-mcgowan

m-mcgowan, 2015-09-10:
Thanks for tracking down the issue to the WKP pin setting - that made it much easier to offer a fix.  :+1:

commit 815e221b (2015-09-10): Set wakeup pin mode when entering standby. Fixes #595