particle-iot/device-os #598: Addition of Wire.reset()

pomplesiegel, 2015-09-09:
We have experienced an issue wherein a MCP23017 port expander gets into a bad state and holds SDA down to ground indefinitely. When this occurs, it seems the only way to purge the bus is to send additional clock pulses down SCL to force the slave to release SDA. See [this AD app note] (http://www.analog.com/media/en/technical-documentation/application-notes/54305147357414AN686_0.pdf) and [this forum post] (http://www.microchip.com/forums/m175368.aspx).
After days of testing, it appears the solution below is quite effective, successfully purging the bus on 7 photons hundreds of times over this period. Would it make sense to include something similar to this as `Wire.reset()` in the Wire library?

````cpp
void RestartI2cBus()
{
  HAL_Pin_Mode(D0, INPUT_PULLUP); //Turn SCA into high impedance input
  HAL_Pin_Mode(D1, OUTPUT); //Turn SCL into a normal GPO 
  HAL_GPIO_Write(D1, HIGH); // Start idle HIGH
  
  //Generate 9 pulses on SCL to tell slave to release the bus
  for(int i=0; i <9; i++)
  {
    HAL_GPIO_Write(D1, LOW);
    delayMicroseconds(100);
    HAL_GPIO_Write(D1, HIGH);
    delayMicroseconds(100);
  }

  //Change SCL to be an input
  HAL_Pin_Mode(D1, INPUT_PULLUP);

  //Start i2c over again
  Wire.begin(); 
  delay(50);
}
````

m-mcgowan, 2015-09-09:
This sounds like a great idea!

pomplesiegel, 2015-09-18:
@m-mcgowan, Awesome! Glad to see this made it to 0.4.6.

closed 2015-09-27 by m-mcgowan