particle-iot/device-os #599: dfu-util: Error during download get_status

m-mcgowan, 2015-09-10:
When using dfu util with the "leave" command, e.g.

```
dfu-util -d 2b04:d006 -a 0 -s 0x80a0000:leave -D tinker.bin
```

The output is confusing since it says the download is successful, and then prints an error:

```
Downloading to address = 0x080a0000, size = 4316
Download        [=========================] 100%         4316 bytes
Download done.
File downloaded successfully
dfu-util: Error during download get_status
```

When used from `make` with `program-dfu` Then we get an additional error from make also:
```
Download done.
File downloaded successfully
dfu-util: Error during download get_status
make[1]: *** [program-dfu] Error 74
```

Further compounding the confusion and uncertainty over whether the file was downloaded or not. :-)

m-mcgowan, 2015-09-10:
After download is complete, the DFU spec prescribes that the device enters a state known as manifestation, where the host gives the device time to apply the download. During this time, dfu-util continually requests the status of the device while it reports that it is manifesting. 

In our case, manifestation isn't required, and the code simply sets the device status to manifestation complete and resets the device. The abrupt reset caused dfu-util to lose the USB connection and report a failure when trying to retrieve the device status.

The main fix is to delay the reset so that the USB response has time to be sent. A secondary fix is to report the state as Download_Idle so that dfu-util doesn't print 

```
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state
```

But simply

```
Download done.
File downloaded successfully
```

With clear messaging, the user is now able to know when the download has succeeded and when it has truly failed.

closed 2015-09-10 by m-mcgowan

monkbroc, 2015-09-27:
Does the bootloader ever get updated for existing development units, or did you do this change for future production units only?

Do you plan on sharing instructions on updating the bootloader to the community? I could write a post with the steps to update the bootloader using the programmer shield (even if that's only doable with special hardware).

m-mcgowan, 2015-09-27:
If you're building from develop then you'll need to build and update the bootloader manually. You can update the bootloader via OTA.

For everyone using the releases, the system will automatically update the bootloader.

monkbroc, 2015-09-27:
OK. Thanks. Good to know.

commit d3cee2ce (2015-09-10): remove the bogus error message from dfu-util in leave mode. fixes #599