particle-iot/device-os #601: const char been corrupted in 0.4.5

JumpMaster, 2015-09-10:
Hi,

After flashing the below code on 0.4.5 it executes first time as expected printing two lists of monthsOfYear.  If the photon is unplugged for 10 minutes upon the next run there is a delay while the time is updated and the second list of months is corrupted.

    #include "application.h"
    
    const char *monthsOfYear[12] = {"January", "February", "March", "April", "May", "June", "July", "August", "September", "October", "November", "December"};
    unsigned long resetTime = 0;
    
    void setup()
    {
      Serial.begin(9600);
    
      for (int i = 0; i < 12; i++)
      {
        Serial.println(monthsOfYear[i]);
      }
      Serial.println();
      Time.zone(1);
    
      do
      {
        resetTime = Time.now();
        delay(10);
      } while (resetTime < 1000000 && millis() < 20000);
    
      for (int i = 0; i < 12; i++)
      {
        Serial.println(monthsOfYear[i]);
      }
    }
    
    void loop()
    {
      Particle.process();
    }

m-mcgowan, 2015-09-10:
Previously the RTC was not being initialized correctly - it was set to a random time, some time in 2013 iirc, so the time returned would be greater than 100000. Now it is initialized to 0, so that it is clear when the RTC has been correctly initialized from the cloud. This gives feature parity with Time behavior on the Core.

JumpMaster, 2015-09-10:
The issue is the const char is been corrupted and the output on a "fresh" boot looks like this

    January
    February
    March
    April
    May
    June
    July
    August
    September
    October
    November
    December
    
    January
    February
    
    
    
    
    Y.
    ¥.
    
    December

m-mcgowan, 2015-09-10:
Wow, that's very strange! Thanks for clarifying - we'll investigate and get back with our findings!

m-mcgowan, 2015-09-10:
I'm still investigating, and this definitely shouldn't happen. 

Just incase you need a workaround, you can ensure that the array ends up in flash, rather than SRAM, by changing the declaration to

```
const char * const monthsOfYear[12] = {"January", "February", "March", "April", "May", "June", "July", "August", "September", "October", "November", "December"};
```

As it was, each string was in flash, but the array itself was in SRAM.

JumpMaster, 2015-09-10:
That's great thanks.  I'm just happy it was reproducible and not just my tools.

JumpMaster, 2015-09-11:
It seems that something is always corrupted.  If I apply the fix you stated a different variable will be corrupted.

Expected output from an HTTP request
    Accept: */*

Actual output
    Accept:
    :                                                                                                                                                                                                                                                  
    :                                                                                                                                                                                                                                               
    ¥.: .
    .

Variable been corrupted
    http_header_t headers[] = {
      { "Accept" , "*/*"},
      { NULL, NULL } // NOTE: Always terminate headers will NULL
    };

jmosk, 2015-09-12:
I noted that v0.4.5 was released earlier today just a few hours after this RAM corruption report. So the question certainly is if the corruption problem has been tracked down and fixed or if the released version may have corruption issues.

m-mcgowan, 2015-09-12:
This is fixed. @JumpMaster please verify and I will close.

closed 2015-09-12 by m-mcgowan

JumpMaster, 2015-09-12:
@m-mcgowan Yes this has fixed the issue  Thanks!
@jmosk Yes this made it into version 0.4.5 which I used to test.

m-mcgowan, 2015-09-12:
Thanks for verifying! It wasn't meant to be closed -  github closed the issue for me automatically from the commit message.

commit e729350c (2015-09-12): setting the unix time calls mallic, which is not allowed in this early startup phase.  fixes #601