particle-iot/device-os #602: serial command for claim code

kennethlimcp, 2015-09-11:
This is related to a discussion about having a photon claimed via Listening mode as a failback from softAP.

(1) get a claim code from the cloud via CLI, 
(2) send the claim code over serial
(3) have the device connect to the cloud

Proposing `c` as the command to be sent over serial

m-mcgowan, 2015-09-13:
I see why you want this, but given that the claim code can already be set via the SoftAP interface, do we need to duplicate this over serial?

With the Photon in listening mode, the cli is able to get the claim code from the cloud and send this to the Photon using the SoftAP interface. The CLI could also use the SoftAP interface to configure WiFi on the photon, which will be more reliable in the long run - the serial interface is intended to be a H2M interface, while the SoftAP TCP/HTTP server is a M2M interface.

kennethlimcp, 2015-09-13:
This allows a user to claim the Photon in an event that the mobile app and CLI claiming via softAP fails. 

We have see it fail for a handful of users and the workflow of claiming becomes very tedious once the preferred softAP method fails. It becomes something like this:

- `particle serial identify`
- `particle serial wifi`
- `particle device add deviceID`
- `particle device rename deviceID deviceName`

Even with all these steps, the Photon did not go through the planned claiming process with claim code.

With the ability to send claim code via serial, we can also verify if the Photon is claimed under another account using the CLI

m-mcgowan, 2015-09-13:
Understood. Could you please specify further how the `c` command would work, in particular how we set the claim code and verify if the photon is claimed?

kennethlimcp, 2015-09-13:
I will have to ask ido about that. The mobile app is the only interface that is currently having this feature but i can check if the CLI has the same feature built-in or not via `particle setup`

m-mcgowan, 2015-09-27:
I clearly wasn't thinking with my previous comment. The softAP API has the ability to set the claim code and determine if a device has been claimed.  I envisage this for serial:

**c**
Device claimed: no
**C**
Enter 63-digit claim code: 04589.....

m-mcgowan, 2016-01-29:
Hey @kennethlimcp! what's the story here - do you still need this?

kennethlimcp, 2016-01-29:
It is used when unboxing fails using CLI. That way we can go through the entire process as though softAP method is used without the user having to retrieve the deviceID, claim and rename.

m-mcgowan, 2016-01-29:
Ok, understood. How does the suggested syntax above look?

@towynlin: is it ok to expose the claim code setting via serial?

kennethlimcp, 2016-01-29:
Sounds ok as long as there's some way communicate the information back and forth between the device and CLI

monkbroc, 2016-04-06:
I support this. When I connect a Photon to USB and do `particle setup` I expect to be able to set it up even if my computer doesn't have a WiFi antenna.

closed 2016-08-09 by technobly

towynlin, 2016-08-09:
Just saw this for the first time, so to close the loop, yes :+1: it's totally fine to allow this over serial. The security requirementfor claiming is that one must have physical access to the device, that is, one must not be able to set the claim code remotely. Thanks much folks!

mebrunet, 2017-02-20:
Is this confirmed to be working? I seem to be having issues using it. Going through the process the device (P1) behaves as expected and confirms the claim code was set. Replying:
```
Claim code set to: <claim_code_I_generated_and_passed>
```
But when it goes online it does not associate itself with the User's account. 

I know the claim code I gave it was good. Because after confirming it has failed via set by serial (e.g. through verification via the dashboard), I can use the claim code in a softap process to successfully claim the device.

m-mcgowan, 2017-09-14:
@mebrunet - I can confirm this is working. Did you find a resolution for your issue?