particle-iot/device-os #604: Photon with SSID out of range, 7 second delay between loop() invocations?

projectgus, 2015-09-11:
I'm moving an application from Spark Core to Photon for a client. They've been using Cores in a situation where the WiFi network is not always present. They use  `SYSTEM_MODE(SEMI_AUTOMATIC)` to run some code immediately after reset, then call Particle.connect() from `setup()`.

The Photon port seems to be working fine (after applying other changes), as long as the WiFi access point is available.

As soon as the WiFi network isn't found, we're seeing a 7 second gap (as measured with `millis()`) between each invocation of `loop()` - I'm guessing as the Photon scans for SSIDs each time around and waits for the response?

On the Core there was no perceptible delay.

Is this a known issue? Is there anything we can do to mitigate it? I've tried building against the current latest branch (0.4.4?) and develop branch (0.4.5rc2?), both seem to have the same behaviour.

m-mcgowan, 2015-09-11:
This is a known issue. I have an app that shows the same behavior. The cause is that the wifi calls are asynchronous on the cc3000 but synchornous on WICED.

The future fix is to use multithreading to run the application on a separate thread. A beta of this will be available next week.

projectgus, 2015-09-14:
Thank you for the update.

We've worked around the issue for now by moving almost all the interactive behaviour to a timer interrupt, but will be eagerly awaiting the threading support so we can undo that!

m-mcgowan, 2015-09-30:
Confirmed that multithreading beta will be released with 0.4.6, which addresses this issue.

closed 2015-09-30 by m-mcgowan