particle-iot/device-os #613: Photon does not connect to wifi with 0.4.5 firmware

Mermaja, 2015-09-14:
Te case is that my photon does not connect to any wifi, neither with the original firmware, 0.4.4 version nor 0.4.5 version.

It goes ok through the set up procedure either by means of the Andorid app or CLI, but once configured it remains blinking green, then solid green, then blinking blue. This happens as ending of the configuration procedure. If I reset the device or power cycle it, it stays blinking green forever.

m-mcgowan, 2015-09-14:
Sorry to hear it's not working for you. Though I'm sure you can appreciate we have thousands of devices that *are* connecting to WiFi, so we need some details about your WiFI setup, the type of router, SSID broadcast/non-broadcast etc will give us something to go on. Have you checked the forum for troubleshooting steps? After trying that you can also file a ticket with our support team.  Filing an issue here is really for when you are sure there is a problem with the device. Thanks for your cooperation and understanding!

JSchaenzle, 2015-09-14:
After updating to 0.4.5 I had the same thing happen when trying to configure it with the Android app. I tried it with the iOS app and it worked the first time. Not sure if that's just coincidence or not but thought it's an interesting data point. I can try configuring it again tonight from the Android app and report back.

Mermaja, 2015-09-14:
Hello.

I have tried several networks, including connecting to an iPhone 4, to the
same result: the photon does not connect. It was the same with previous
versions of the firmware.

The router that I am using now is a Buffalo WHR-HP-G54 implementing the
IEEE 802.11g standard. The SSID is broadcast and the network has no
password.

As administrative information, I come here from the forum and an advise for
a member of Particle support, that suggested to make an issue here.

Regards.

m-mcgowan, 2015-09-14:
Thanks for the additional details. Have you tried configuring wifi via particle CLI? If so, did that work or fail?

Mermaja, 2015-09-14:
I have tried twice, once to connect to my wifi and the other to connect to
an iPhone. The procedure goes OK, but when ending the photon goes blinking
green, then solid green, then blinking blue. It does not connect at all. If
I press reset or cycle the power, the photon only blinks green, and does
not connect.

m-mcgowan, 2015-09-27:
Any change with this? I'm not sure what to suggest since I don't see anything specifically different with your setup.

Mermaja, 2015-09-28:
No news regarding my photon. I have tried many times, many networks, and 
it refuses to connect. It is weird as if it is able to detect the SSIDs 
of the surrounding networks, if I can connect to it using its AP mode to 
configure, I presume it is not a harwdware issue. But on the other hand, 
the system does not connect. May the problem be permanent corruption on 
the configuration storage memory? I am almos sure that it is not a 
software issue.

Thank you for your interest.
Regards.
Germán Fabregat.

m-mcgowan, 2015-09-28:
Here's what I'd like you to try.

First, erase all existing wifi credentials by holding the setup button down for 10 seconds. You'll see the LED first blink blue, then after 10 seconds blink fast blue. The wifi credentials are then cleared.

Update your CLI to the latest version using `npm install -g particle-cli`

Run `particle setup wifi' to setup your wifi network. Choose 'n' at the first prompt to not scan for networks. 

Please paste the output of particle cli here, including all the prompts. Thanks!

Mermaja, 2015-09-28:
Here you are. As furter information, the photon has not restarted, it 
remains blinking blue. I have pressed reset and is solid green.

Microsoft Windows [Versión 6.3.9600]
(c) 2013 Microsoft Corporation. Todos los derechos reservados.

C:\Users\G>npm install -g  particle-cli
-
> serialport@1.7.4 install C:\Users\G\AppData\Roaming\npm\node_modules\particle-
cli\node_modules\serialport
> node-pre-gyp install --fallback-to-build

[serialport] Success: "C:\Users\G\AppData\Roaming\npm\node_modules\particle-cli\
node_modules\serialport\build\serialport\v1.7.4\Release\node-v14-win32-x64\seria
lport.node" is installed via remote
C:\Users\G\AppData\Roaming\npm\particle -> C:\Users\G\AppData\Roaming\npm\node_m
odules\particle-cli\bin\particle.js
particle-cli@1.8.3 C:\Users\G\AppData\Roaming\npm\node_modules\particle-cli
├── fast-luhn@1.0.2
├── cli-spinner@0.2.1
├── xtend@4.0.0
├── when@3.7.3
├── chalk@1.1.1 (supports-color@2.0.0, escape-string-regexp@1.0.3, ansi-styles@2
.1.0, has-ansi@2.0.0, strip-ansi@3.0.0)
├── node-wifiscanner2@1.1.0 (os-locale@1.4.0)
├── cli-table@0.3.1 (colors@1.0.3)
├── hogan.js@2.0.0
├── glob@4.5.3 (inherits@2.0.1, once@1.3.2, inflight@1.0.4, minimatch@2.0.10)
├── moment@2.10.6
├── request@2.64.0 (aws-sign2@0.5.0, forever-agent@0.6.1, stringstream@0.0.4, ca
seless@0.11.0, tunnel-agent@0.4.1, oauth-sign@0.8.0, isstream@0.1.2, json-string
ify-safe@5.0.1, extend@3.0.0, node-uuid@1.4.3, combined-stream@1.0.5, qs@5.1.0,
form-data@1.0.0-rc3, http-signature@0.11.0, mime-types@2.1.7, tough-cookie@2.0.0
, bl@1.0.0, hawk@3.1.0, har-validator@1.8.0)
├── lodash@3.10.1
├── softap-setup@1.1.4 (nconf@0.7.2, node-rsa@0.2.25)
├── inquirer@0.8.5 (ansi-regex@1.1.1, cli-width@1.0.1, figures@1.4.0, through@2.
3.8, readline2@0.1.1, rx@2.5.3)
└── serialport@1.7.4 (bindings@1.2.1, sf@0.1.7, async@0.9.0, nan@1.8.4, debug@2.
2.0, optimist@0.6.1)

C:\Users\G>particle setup wifi
? Should I scan for nearby Wi-Fi networks? (Y/n) n
? Should I scan for nearby Wi-Fi networks? No
? SSID: mermaja
? SSID: mermaja
? Security Type: WEP
? Wi-Fi Password: lavid
Done! Your device should now restart.

C:\Users\G>

m-mcgowan, 2015-09-28:
Right You didn't mention it was WEP. For a WEP network you need to configure the passkey in a special way. I can tell you how to do this, but I would suggest instead you switch to WPA or WPA2, is that possible?

Mermaja, 2015-09-28:
This network is WEP, but I have tried with non protected ones, WPA and 
WPA2 to the same result.

If you need it for the sake of your debugging, I can try to connect to 
my iPhone's network, no problem at all with doing it. I will send you 
the results in a moment.

Mermaja, 2015-09-28:
I have just tried 'particle setup wifi' as I presume CLI is up to date. 
Now it has shown a new behaviour, as the photon did reboot, it has 
flased red for several seconds, then flashed green, where it remains 
-with the iphone sharing the network and showing no device connected.

I attach the output.

C:\Users\G>particle setup wifi
? Should I scan for nearby Wi-Fi networks? No
? SSID: iPhone Mermaja
? SSID: iPhone Mermaja
? Security Type: WPA2
? Wi-Fi Password: opals0952
Done! Your device should now restart.

C:\Users\G>

m-mcgowan, 2015-09-28:
Can you run `particle --version`

m-mcgowan, 2015-09-28:
I'm not sure about connecting to iPhone hotspots - can you instead reconfigure your router to use WPA2? Alternatively you can follow the guide here to correctly format a WEP password - https://community.particle.io/t/how-to-setup-wep-secured-network-on-the-photon/16026

Mermaja, 2015-09-28:
1.8.3

Mermaja, 2015-09-28:
Hello.

I was advise some weeks ago by someone from Particle to test against my 
iPhone as it tends to be more "well-known" wifis.

I have tested again my access point using the correct WEP settings, and 
the behaviour is more as expected, but it does not connect. Now the 
photon rebooted, it flashed green, then solid green, then blue and it 
remains there.

As lavid was the first key and its size is 5, I have entered as key, 
following the link you gave me:

00056c61766964

I have tried twice -from the beginning, that is erasing the 
configuration between tries- both with capital an low c.

m-mcgowan, 2015-09-28:
Thank you for trying these things. Other than changing the router to WPA2 I don't have any further steps to suggest. We will be releasing 0.4.6 in a couple of days, that may help.

Mermaja, 2015-09-28:
Thank you very much. Please understand that if I do not change my router 
settings is because I have several devices connected and I do not want 
to reconfigure them. And, more important, because I have also tried with 
other WPA2 routers with the same frustrating results.

I own a Core that is working OK and I am looking forward an Electron 
that I require for a future application. I bought the photon just for 
playing with it, so I am not in a hurry.

Regards.

Mermaja, 2015-09-28:
Hello again.

I have been thinking about the problem. I develop hardware and software, 
in fact I teach computer architecture and electronics, so I have a 
certain degree of knowledge on the subject. Imagine that the memory 
where the device stores the configuration -SSID and key- is damaged. 
Apparently everything goes ok but when reading the stored data the 
photon gets wrong values and so it is not able to connect.

Is this possible and reasonable? Is there any way to read and show via 
CLI these settings?

Thank you for your support.

m-mcgowan, 2016-01-29:
The 0.4.9 firmware release contains a new API - `WiFi.getCredentials()`. You could use that in your app to output the current configured credentials if you wanted to test your theory about memory corruption.

You can get the release binaries from here - https://github.com/spark/firmware/releases/tag/v0.4.9-rc.3. Please let me know how it goes! :-)

Mermaja, 2016-01-29:
Hello.

Thank you for the info. It arrives at te best moment as I am planning to 
use my photons for some projects with my students.

I will keep you informed.

Regards.

Mermaja, 2016-02-01:
Hello.

The Photon behaves the same way, in spite of the new firmware. I flashed 
it, configured my WIFI but it did never connect. I also tested my other 
2 Photons, and they connected without problems.

Thank you again for your help.

Germán.

m-mcgowan, 2016-03-14:
Did you get your device to connect? I suggest posting to our forums at http://community.particle.io where you'll get many more eyes on the problem!

Mermaja, 2016-03-15:
Hello.

Thank you for tour advise, but I have been posting and talking to the 
firmware development team, and nothing has changed :(

I have tried the three or four last firmware versions to the same 
results. In addition Particle kindly sent me another Photon that worked 
without problems, but I will be glad of performing more tests or sending 
the damaged one back to Particle if requested.

Regards,
Germán Fabregat.

m-mcgowan, 2016-03-15:
Thanks for the confirmation. It sounds like it could have been a hardware issue. I'll close this issue for now. If you have an external antenna you might try that (and check WiFi.selectAntenna() documentation.)

closed 2016-03-15 by m-mcgowan