particle-iot/device-os #620: Setting up Wi-Fi via serial sometimes locks up

zsup, 2015-09-16:
Steps to reproduce:

- Open a new Photon (factory firmware)
- Use `particle update` to update to v0.4.5 using CLI v1.8.3
- Type `screen /dev/tty.usbmodemXXXX` to open serial terminal
- Hit `w`
- Set up Wi-Fi credentials (I used a secured WPA2 network)

I'm setting up a series of devices like this, and about half the time, it starts flashing green, then the green LED stays on for about five seconds, and the device reverts to flashing blue. A reset gets the device offline.

The other half of the time it works as expected - flashing green, flashing cyan, breathing cyan.

m-mcgowan, 2015-09-16:
Thanks for this - always great to get reproducible steps! 

When the device goes back to listening mode (flashing blue) when in listening mode, it means it couldn't connect to the Access Point.  

- Could it be a typo in the password? 
- Perhaps the connection timeout is too short, not giving the device enough time to join the network.

After entering the SSID the device scans for the network to determine the security cipher (AES/TKIP). If it doesn't find the network it will prompt for the security cipher, since it's assuming the network is hidden/offline.  If you don't get this prompt, then it means the device has found the network, so the reason for not connecting is most likely one of the two things above.

> A reset gets the device offline.

So the device stays offline (flashing green) after a reset?

zsup, 2015-09-16:
Not a typo, has happened a number of times, and a reset gets it to connect successfully, so it had the correct credentials.

After a reset, the device connects successfully (flashing green, flashing cyan, breathing cyan).

zsup, 2015-09-16:
Also the flashing green only lasts for one or two seconds before it hits solid green, if that is helpful re: your second theory (short timeout).

zsup, 2015-09-16:
Here's the serial output:

```
SSID: ThingLabs
Security 0=unsecured, 1=WEP, 2=WPA, 3=WPA2: 3
Password: ThingLabs
Thanks! Wait while I save those credentials...

Awesome. Now we'll connect!

If you see a pulsing cyan light, your device
has connected to the Cloud and is ready to go!

If your LED flashes red or you encounter any other problems,
visit https://www.spark.io/support to debug.

    Spark <3 you!
```

karlhenricksen, 2015-09-28:
@m-mcgowan , I'm seeing the same behavior when setting up Wifi creds over the air - occasionally the photon seems to give up and revert to the listening mode (blue blinking light). I've seen this on an Netgear N router, WP2/AES, about 10% of the time. I am doing software QA work on a Photon project at the moment, and hoping to get an idea of the scope of wifi setup issues. 

I thought I had this nailed down to fully reproducible steps when I was using a B-only Linksys wrt-54 router, but then it was successful multiple times after and haven't reproduced again consistently. But it sounds plausible that we just may be on the cusp of the timeout at certain times. 

Any other insight into this issue though? Other factors that may cause a longer time to connect to test that theory? I'll be testing in this area this week, so may be able to give more info with steps to reproduce.

m-mcgowan, 2015-09-30:
I have increased the number of retries from 1 to 3 before giving up. each retry takes 7 seconds.

m-mcgowan, 2015-09-30:
Gong to close this now, assuming fixed with the 0.4.6 release. Happy to reopen if the problem reappears.

closed 2015-09-30 by m-mcgowan