particle-iot/device-os #635: SoftAP HTTP Server Missing `Access-Control-Allow-Origin: *`

msolters, 2015-09-22:
While GET requests to the SoftAP HTTP server are handled fine, it seems that POST requests are missing the `Access-Control-Allow-Origin: *` header.  This causes some browsers, like Safari, to fail at the `configure-ap` step of browser-based SoftAP.  See below.

[image]

[See community post for more info.](http://community.particle.io/t/photon-softap-usage/11035/150?u=msolters)

Now, I see the following macros inside `http_server.h`:

```c
#define HTTP_HEADER_CONTENT_LENGTH        "Content-Length: "
#define HTTP_HEADER_CONTENT_TYPE          "Content-Type: "
#define HTTP_HEADER_CHUNKED               "Transfer-Encoding: chunked"
#define HTTP_HEADER_LOCATION              "Location: "
#define HTTP_HEADER_ACCEPT                "Accept: "
#define HTTP_HEADER_KEEP_ALIVE            "Connection: Keep-Alive"
#define HTTP_HEADER_XHR					  "Access-Control-Allow-Origin: *"
#define HTTP_HEADER_CLOSE                 "Connection: close"
```

`HTTP_HEADER_XHR` is exactly what we want the HTTP server to be stamping on *all* resource methods - GET and POST.

But I fail to see where this header -- or indeed any of these headers -- are ever actually used.  Where is the WICED HTTP server configured??

m-mcgowan, 2015-09-22:
Which version of firmware are you using?

The WICED code is private due to licensing restrictions, the relevant part added by us is:

```
    if (stream->cross_host_requests_enabled == WICED_TRUE)
    {
        WICED_VERIFY( wiced_tcp_stream_write( &stream->tcp_stream, HTTP_HEADER_XHR, strlen( HTTP_HEADER_XHR ) ) );
        WICED_VERIFY( wiced_tcp_stream_write( &stream->tcp_stream, CRLF, strlen( CRLF ) ) );
    }
```

This is activated in this commit - https://github.com/spark/firmware/commit/2c98d1c7a477c8963e9ebfe154887fb807e35481

You'll need version 0.4.5 of firmware to have this commit on your device.

msolters, 2015-09-22:
The errors above happen to 0.4.5 as compiled by the cloud, as well as if one were to build `develop` branch locally.

m-mcgowan, 2015-09-22:
Afaik POST is an accepted HTTP method. Do you have any more details on when it works/doesn't work?

msolters, 2015-09-22:
Yes.  The [thread I linked to earlier](http://community.particle.io/t/photon-softap-usage/11035/) has plenty of examples where people using Safari have found that the `configure-ap` command, a POST request, just hangs forever, while all GET requests up until that point (`/device-id`, `/public-key`, `/scan-ap`) work fine.

And yet, in Firefox & Chrome, it usually works!

What I've been doing is systematically forking `softap-setup-js` and using a bevy of different request handlers -- `http-browserify`, `http-browserify-xdr`, `xhr`, etc.

From what I can gather, based on a user's browser and its stringency in enforcing CORS security, different JavaScript libs fall back to different request handlers to attempt to maximize compatibility.

I'm trying some more variations now.  There is certainly a JS aspect to this problem, but I can't get around the fact that *something is not quite right* with the WICED http server as written.  Even Chrome and Firefox will fail to connect with XHR.

m-mcgowan, 2015-09-22:
If you could produce a text file containing an HTTP request that fails,  I can then netcat to the HTTP server that will help me debug the issue.

msolters, 2015-09-22:
The following information would help:

*  Inside the WICED HTTP server, what content types are permitted for the `configure-ap` route?
*  What kind of response header are we expecting?
*  Is OPTIONS considered an HTTP verb, as configured?

As far as an HTTP request that fails, the above image contains precisely that information on the lower RHS.

msolters, 2015-09-22:
In addition, if I make the `configure-ap` request as  GET, I actually do get a response from the Photon's HTTP server: `{r: 0}`.  But the Photon doesn't receive any payload in that case.

Here's a far more concrete example.  [Use this site in Safari](http://photonsoftap.meteor.com), and check out Network pane of the dev console.  You'll see precisely what gets sent to the Photon -- but the Photon never resolves the request!

anthonywebb, 2015-09-22:
@msolters I am able to get a little further, in safari I actually make a successful POST to configure-ap, that I then get a response back ({r: 0}).  It does not, however actually save my posted creds unless I do it from chrome.  You may want to play with the headers you send in your POST.  See the method "postRequest" near the bottom of the JS.

http://23.253.54.117/setup
http://23.253.54.117/setup/script.js

[image]

msolters, 2015-09-22:
Hey @anthonywebb, that's exactly why I asked this question:

> Inside the WICED HTTP server, what content types are permitted for the configure-ap route?

I know considerations are being made in the server, as they must be, because I see these macros are defined:

```
#define HTTP_HEADER_CONTENT_LENGTH        "Content-Length: "
#define HTTP_HEADER_CONTENT_TYPE          "Content-Type: "
#define HTTP_HEADER_CHUNKED               "Transfer-Encoding: chunked"
#define HTTP_HEADER_LOCATION              "Location: "
#define HTTP_HEADER_ACCEPT                "Accept: "
#define HTTP_HEADER_KEEP_ALIVE            "Connection: Keep-Alive"
#define HTTP_HEADER_XHR					  "Access-Control-Allow-Origin: *"
#define HTTP_HEADER_CLOSE                 "Connection: close"
````

My guess is that we are somehow -- and Chrome/Firefox perhaps guess and fill these in -- not passing valid Content-Type or some other such parameter, so that the HTTP server either doesn't recognize the request, or recognizes it but can't decode it.

Without looking at the WICED logic this is really hard to pinpoint.

anthonywebb, 2015-09-22:
As you noted, 1) sometimes the server will not allow a POST as it should, and my issue 2) when proper headers are set such that the POST is accepted, the data posted is not retained properly.  

I guess I am wondering if the issues are related or not.  I think that if the server is accepting the POST with a  200OK and a response, but never retaining those credentials, then that is a firmware bugs that needs to be noted/resolved.  By my calculation, Issue 1 can be resolved by sending the proper headers in your POST, right?

msolters, 2015-09-22:
Yeah, I would agree.  I want to look at the HTTP server so I can know exactly what headers have to be set, and what AREN'T set explicitly -- because different JS libs on different browsers will choose different defaults.

msolters, 2015-09-22:
Ok, I have a version which works in Safari and Chrome.  Here's the headers that you need:

```
{
  'Content-Type': 'multipart/form-data',
  'Accept': '*/*',
  'Accept-Encoding': 'gzip, deflate'
}
```

Safari will probably throw a fit that you dared to set the Accept-Encoding, but it *will* work, and it *will* retain the values. :)

closed 2015-09-22 by msolters

m-mcgowan, 2015-09-22:
Just to clarify, you have this working and no need for firmware changes?

msolters, 2015-09-22:
*Sort of.*  It requires a huge number of assumptions in browser land because we can't see the source code -- and so portions of what I'm doing might be overkill.

But, yes, I have it working in Safari now.  I'll need to do more testing before boiling this down to the core minimum request headers that are required.  In addition, it seems that the configuration required means the SoftAP JS lib may have to be rewritten on the hTTP side.  But I know you are busy with other problems so consider this closed.

anthonywebb, 2015-09-22:
@msolters my hats off to you!  I am trying to duplicate your success but am still seeing that my wifi creds are not stored in (Mac) Safari 9.0 (10601.1.56), I tried on my iPhone (iOS8) with the same results

My code to send the post looks like:
```
  var xmlhttp = new XMLHttpRequest();
  xmlhttp.open('POST', url, true);
  xmlhttp.timeout = 4000;
  xmlhttp.setRequestHeader('Content-Type', 'multipart/form-data');
  xmlhttp.setRequestHeader('Accept', '*/*');
  xmlhttp.setRequestHeader('Accept-Encoding', 'gzip, deflate');
  xmlhttp.send(dataString);
```

msolters, 2015-09-22:
Yeah here's the fucking weird thing, man:  all browsers will throw an error saying they refuse to set the `Accept-Encoding` as its an unsafe header.  But if you *don't explicitly set it*, the Photon won't take your credentials from Safari.  WTF.  Safari in a nutshell right there.

As to getting this to actually work, I had to migrate SoftAPSetup to an entirely different node module, [XHR](https://www.npmjs.com/package/xhr), to handle HTTP.

It seemed like each successive tool that browserifies CORS type requests is sort of an improvement in the exercise of abstraction in some way, and of course Safari is not about improvements.  So I went with the package that says "Browser support: IE8+ and everything else." right in the description.  Seems to work!  Let me clean this up and sanitize it a little bit, I will put a PR up against `spark/softap-setup-js` so you can see what I did.

msolters, 2015-09-22:
Ok so in fact here's my current state of affairs:

*  Open networks *always* work - the Photon retains the credentials and then connects.
*  Secure networks are a toss up.  Sometimes they take.  Sometimes they don't.

I am suspicious of this `multipart/form-data`.  What does WICED's HTTP server expect?  I just wish I knew!

m-mcgowan, 2015-09-22:
I don't think WICED cares about the content type, it just forwards it to the application handler.

msolters, 2015-09-22:
No, this is demonstrably false.  Failing to explicitly override the content type with `'Content-Type': 'multipart/form-data'` causes the Photon to completely ignore otherwise valid network requests from Safari.  Content type is clearly an important factor in this mechanism.  Specifying the content type then causes the Photon to recognize and process the payload -- as long as there is no password.  That is another problem.

msolters, 2015-09-22:
Content-Type of Request (Safari) | Photon Behaviour
---|---
`multipart/form-data` | Photon connects to WiFi, but only open WiFi. With secure networks, Photon seems to receive something (blue blinking is interrupted for a second) but then resumes listening mode.  On reboot, enters listening mode.
`json/application` | `configure-ap` hangs indefinitely for both open and secure networks.
none (`application/x-www-form-urlencoded`) | `configure-ap` hangs indefinitely for both open and secure networks