particle-iot/device-os #640: [core] support for wakeup pin

m-mcgowan, 2015-09-25:
The wakepup pin support was never released, and was removed here since it caused A0 to not function as expected.  https://github.com/spark/firmware/issues/595

Wakeup on the core is a useful feature, so let's investigate what it takes to get this working!

schwiet, 2016-01-06:
are these bugs related (#655)? did wakeup on the core ever get supported? It seems to work intermittently for me, but not reliable. Thanks.

m-mcgowan, 2016-01-06:
I don't believe they are related. The WKP pin on the Core functions differently to how the EXTI interrupts are handled with stop mode.  Please see issue #595 for why we dropped the attempted and unreleased implementation of WKP on the Core. 

If you, or anyone else, is able to investigate why the wakeup code caused A0 to malfunction that would be very much welcomed!

schwiet, 2016-01-06:
Ok, thanks for the reply. So just to be sure I understand, `System.sleep( pin, mode )` is not supported on the Core? The documentation does imply that this should function properly using an updated boot loader.

m-mcgowan, 2016-01-23:
`System.sleep(SLEEP_MODE_DEEP, time)` isn't supported since the wakeup pin (A0) isn't configured as wakeup on the Core. 

`System.sleep(pin,mode,duration)` works, where duration can be 0 for no timed wakeup.

closed 2020-02-28 by avtolstoy